samphonic/Moppy2 #1: Allow toggle of head-movement via MIDI message

SammyIAm, 2017-11-14:
Allow toggling of floppy drive head movement via a MIDI message.  In two parts:
* A MoppyMessage command to set the maxPosition value for a subAddress (useful also for remote configuration)
* A MIDI message to trigger the above message.  Suggest using the "Control Change" message type.  This message would be interpreted to set the maxPosition either to "2" (which would lock the head in place), or setting maxPosition to... whatever it was before?

Additional note: Setting to "2" would actually cause the head to move back to start and *then* stay in place, so maybe we need a separate toggle value on the Arduino side.

Lothean, 2018-07-16:
The MoppyMessage could also allow us to configure the maxPosition from the GUI, and could lead us to a new user experience: everything could be configured directly from the software!

SammyIAm, 2018-07-19:
Yeah, that's definitely occurred to me as a future improvement, but it's a bit daunting since it would require a fair bit of messaging infrastructure (plus you'd want the settings to be stored in the EEPROM, so that's an extra bit of code).  Makes sense to put in that effort though with the addition of support for this issue, so it's being considered.

Lothean, 2018-07-21:
Would be better to store the settings in the Java side and transmetting "init" commands at the start ?

SammyIAm, 2018-07-23:
It would be less work to *only* send an init, but you wouldn't be able to read the current state of the Arduino, and any time the Arduino reset it would need to be re-inited (which means a separate new message for the Arduino to request init that only gets sent once-- but you'd have to make sure the app was up and running and connected first or it would miss the message or......)

At the end of the day it'll be the same amount of trouble to just do it the right way the first time around :P

SammyIAm, 2018-07-23:
According to ATMEL's (*very* conservative from the sounds of it) estimate of 100,000 cycles for the Arduino EEPROM, if you changed a setting *every* 5 minutes for 4 hours a day (gotta take breaks from Moppy for adulting, you know 😉) *every single day*, then you'd expect to *maybe* see a failure after about 6 to 9 years.  Some [real-world testing](https://hackaday.com/2011/05/16/destroying-an-arduinos-eeprom/) seems to suggest that you could probably keep it up for about 70 years before you saw an error.

So while *technically* a concern, no one's likely to be changing settings quite that often.

-- Actually, I guess I'm getting a bit off of the original intent of this issue which was to have a MIDI message change the head-movement.  If it's something that can be changed via MIDI messages, EEPROM is a poor choice (you'll potentially be getting lots of MIDI messages), but for anything GUI/settings related I think it should probably suffice.

Sammy1Am, 2020-08-16:
Added by commit 12901a31a394cefd11f815fde511e42d0585e5d2

closed 2020-08-16 by Sammy1Am