samphonic/Moppy2 #6: MIDI Input and Output support

SammyIAm, 2018-05-04:
Add the ability for MoppyControl to accept MIDI events from an MIDI In source instead of a midi file, and the ability to add MIDI Out ports as a "device".  (This might require converting MoppyMessages back into MIDI Messages since much of the mapping and routing relies on the MoppyMessage format).

Chong-McBong, 2018-05-14:
thanks for this, just in time for my (in progress) midi drum robot, and midi dot matrix printer builds :)

SammyIAm, 2018-05-14:
Hey, so, I have a question for you if you're interested in this feature.  The MIDI _In_ is no problem, it basically just replaces or adds to the sequencer we're already using.  For MIDI _Out_ there are two options:
1. MIDI events are converted to MoppyMessages, run through the mapping scripts like normal, and then get converted back into MIDI events on output.  MIDI Out would essentially show up as "device" connected to the computer.  This would let you send only certain channels / notes to the MIDI Out, but would throw away any MIDI events not convertible to MoppyMessages (control change, &c.).
2. MIDI events are piped directly from MIDI In (or a MIDI file) right out to MIDI Out.  This means that *all* the MIDI events will get sent to the MIDI Out port so if you wanted only some channels played you'd have to do the filtering on the receiving end, but you'd get every single MIDI event in case you needed them.

Both options would allow you to sync MIDI devices to the Floppy Drives, it's more a question of if you need all the events and where you're going to do the channel filtering.  Technically it's also possible to do *both* and figure out a way to encapsulate arbitrary MIDI event data into special MoppyMessages for routing and unbox them on their way to MIDI Out, but that's a bit more complicated.

Chong-McBong, 2018-05-14:
for midi out, option 2 sounds the best i think. 

my drum robot is built to recieve raw midi, then it listens for data on channel 10, octave 2, and hits corresponding drums (i will have to edit the midi files to align the drums with my setup, but it should work in theory) 

i plan to do the same for the printer, hopefully i can listen for data for a midi channel (the same way as the drum robot) then split (using yet another arduino) to the 24 dot-matrix pins 

this way i get 24 note polyphony on only 1 midi channel leaving the rest for other instruments- might be more efficient than trying to do more than 16 channels :)

i don't know if i need all the other events, but they might be useful for people using synths and stuff

Chong-McBong, 2018-05-14:
excellent, thanks 👍 

unfortunately i'm completely allergic to writing code, but i have the hardware mostly complete, going to bash my head against the keyboard this week and see if i can get something working

i will look into using an instrument class, but i feel a steep steep learning curve ahead, so it might be a while before i have any progress on the printer. drums are almost working, so i will do them first i think

DJthefirst, 2018-05-22:
I agree my Tesla controller searches for the Chanel and can play two devices on the one board, so the multiple channels would be useful. Also would you be able to use fl studio or mixcraft midi outputs to put into moppy directly. I tend to have to constantly move back and forth editing midi files when they have direct outputs.

SammyIAm, 2018-05-22:
> would you be able to use fl studio or mixcraft midi outputs to put into moppy directly[?]
Yup!  You might need to get a utility to act as a virtual MIDI In/Out, but you would be able to do that.

VegaDeftwing, 2018-06-01:
I also agree with 2 since it's the most 'complete' option and offers more freedom, even if it's at some expense of ease of use.

Also, heads up, you might want to be careful about pitch adjustment in the live source. Also, adding support for the mod wheel to do something cool might be interesting, say, controlling the number of outputs to use?

SammyIAm, 2018-06-01:
Pitch bending is supported currently, and configured in MoppyInstrument.h to 200cents at full deflection by default.

Mod wheel support could definitely be added by creating a new MoppyMessage type to contain the events.  I haven't come across a compelling use case yet, but I've been intentionally leaving that door open.

SammyIAm, 2018-06-04:
Alrighty, just added a commit that adds:
* **MIDI In**: Messages from this MIDI device will behave the same as if they'd come from the sequencer (i.e. be sent through the mappers and on to the floppy drives, etc.)
* **MIDI Out**: Messages from the sequencer and any selected MIDI In device will be piped straight through to this device (by passing any mapping, etc.)

Haven't built a release yet so you'll have to build your own from source-- if that's an issue for anyone here let me know and I'll make a release specifically for MIDI support.  Otherwise, definitely let me know how it works for you and please open a new issue for any problems you run into.

closed 2018-06-04 by SammyIAm