samphonic/Moppy2 #8: Not mentioned about Serial library

redcubie, 2018-05-11:
The Serial library you need is at [http://fazecast.github.io/jSerialComm/binaries/jSerialComm-2.0.2.jar](url).

SammyIAm, 2018-05-11:
Gradle should be automatically pulling that library down from Maven for you:
https://github.com/SammyIAm/Moppy2/blob/master/Java/MoppyLib/build.gradle#L33

If it's not doing so automatically, there may be an issue with either the Gradle config of the project, or the Gradle config of your Eclipse plugin.

SammyIAm, 2018-05-12:
Yeah, I've a bit of a like-hate relationship with Lombok myself but when it stays out of the way it's handy to have.

I've added the information you've provided on importing the project to the dev wiki page, if there's anything else that might be helpful to add you can either put in in this or another ticket or I guess there's a way to clone the wiki itself as a git repo and do a pull request:
https://github.com/SammyIAm/Moppy2/wiki/Getting-Started-with-Moppy-Development

SammyIAm, 2018-05-24:
Sounds like you got the serial drivers working so I'm going to close this for now.  Feel free to reopen if there's still serial issues, or create a new issue if there's something else not working.

closed 2018-05-24 by SammyIAm