samphonic/Moppy2 #12: Toccata and Fugue in D Minor midi?

cybertofer, 2018-05-27:
hi sir, my name is tof from belgium, i m a music lover and i love to plug electronics to stuff and make them sound! i followed your tutorial and i have now a bunch of floppy who can sings all together. i m using your soft , midi yoke as a bridge and synthesia to play the songs, the result is really cool; thanks for your tutorial and sharing! i would like to ask you, if you agree, the midi file of Toccata and Fugue in D Minor you played on youtube if you dont mind. i reallly like to see it on my floppy orchestra . thanks a lot for your job, tof.

SammyIAm, 2018-05-27:
Hello there!  For various IP reasons I'm trying to avoid just posting the MIDI files anywhere, but I just sent an email to your hotmail.com account with the file attached.  Let me know if it didn't arrive.

FYI: MIDI In support should be coming soonish (when I have time) for Moppy 2.0 so you can switch from the previous version. 😄

closed 2018-05-27 by SammyIAm

bonnevillemariner, 2018-10-29:
Any way I can get this file too? Just built my "organ" and I'd love to teach my students how to work it with this midi!

cybertofer, 2018-10-29:
hello sammy, wich file do you need? the toccata ne you sent me?

bonnevillemariner, 2018-10-29:
Yes, Toccata and Fugue, if you don't mind.

bonnevillemariner, 2018-10-29:
Would you mind sending it to bonnevillemariner @ gmail?