samphonic/Moppy2 #13: Apple Floppies

DJthefirst, 2018-06-06:
I’ve found some old apple floppies and they look like they have serial wires and pinnouts. I found some diagrams showing their is a pwm pin but no direction pin. is their a way anyone else knows how to drive them with moppy. (I’d rater not just drive the pwm signal to the motor as that just plays like as a speaker )

Lothean, 2018-06-06:
Hello
As far as I know, Apple drives are not compatible with Moppy, sorry

SammyIAm, 2018-06-07:
Any chance you could share the pinouts/diagrams here?  From my little research it seems like they definitely don't have the same super-convenient step/direction pins that the rest of floppies have, but it doesn't mean there's not some way around it.

(But out of the box, @Lothean 's right, probably not going to work)

DJthefirst, 2018-06-07:
http://old.pinouts.ru/HD/MacExtDrive_pinout.shtml
Here is the link for the page I found it on

DJthefirst, 2018-06-07:
If you want me to modify the drive inside I can I have a 5” a 3.5” and another 5” mit drive but that has normal pin outs I’d send pics but I don’t think I can from my phone.

SammyIAm, 2018-06-07:
That's a better pinout than I'd found, but confirms my suspicions: rather than controlling the steps and direction with pins, the control line(s) are used to send commands and the drive handles the steps and direction on its own.  I suspect the PWM to regulate speed might be more about regulating the speed the drive reads and writes bits to send over the data lines than the physical speed of any motors.

Because you can't use the pins on the drive interface, your next best option will be to drive the stepper motor directly with a stepper motor controller.  You'll need to do some (de)wiring in the drive, but stepper motor controllers should be reasonably compatible with Moppy.

DJthefirst, 2018-06-07:
It is a 6 pin floppy I’ll see if I can trace back the traces to the chip and if it has a pwm and step pin

Lothean, 2018-06-08:
If it's a 6 pin stepper motor, it _should_ be compatible with the DRV8825 or the A4988.
The input voltage of these does not matter, just set the current very low to start and see if the motor gets too hot.
These previsously mentionned board are cheap, not hard to use for beginners like me and have the convenient STEP/DIR interface.

DJthefirst, 2018-06-12:
The part I find weird is if I trace it back it goes to an ba12003 buffer and then out to the input. So it means it the stepper was driven by the computer?

SammyIAm, 2018-06-12:
Hmm, that does seem weird.  On the one hand, "normal" floppies have their stepper motors sort of driven by the computer in that it says when to step (so it's not too far off from that), but on the other it seems weird that the computer would also need to track the current state of the motor (that seems more like an IC thing).

Lothean, 2018-06-12:
http://www.datasheetlib.com/datasheet/733481/ba12003_rohm.html
If it's the chip in question, it's a transistor array 
You need to go further in the research, give us details about where everything is connected ?