samphonic/Moppy2 #14: Disable/enable reset after track ends

Lothean, 2018-06-12:
Hey, 
Don't know about you, but the auto reset lacks a "checkbox" feature, or like in the first Moppy, with delay and all
Serisously, that's sooooo annoying, it messes up all my audio recordings and my scanners positions (since the L298N does not wind back the scanner to the beginning when resetting for obvious reasons)
Could someone consider making this? I could try but I'm bad at Java and I'd rather not mess up the clean code ;)

SammyIAm, 2018-06-12:
As an interim solution you can just remove `MoppyMessage.SYS_RESET` from [this line](https://github.com/SammyIAm/Moppy2/blob/6d56865e66fdc7b9b81f48336fd6a4b5d06dd99f/Java/MoppyLib/src/main/java/com/moppy/core/events/mapper/BaseEventMapper.java#L25) to disable resetting.  There'll be no manual reset option aside from resetting the Arduino, but it won't ruin your recordings.

> the L298N does not wind back the scanner to the beginning when resetting for obvious reasons

They may be less obvious to me-- out of curiosity, why doesn't it move the scanner back to the beginning?  And how do you get it back to the beginning?

Lothean, 2018-06-12:
Hey!
Because of the high number of steps, moving the scanner in the same manner as a floppy drive would cause it to jam during a long time against the chassis and this repetitive stress could cause damage to the gears/belts of the scanner, and the sound is also very annoying ;)
To move the scanner back to the beginning, I just unplug the power and push it back manually!

SammyIAm, 2018-06-12:
Ah, makes sense.  Perhaps a modification to at least step back until the `currentPosition` is 0 would be in order.  As long as the L298N doesn't drop any steps, that would put it right back at the beginning (and worst case would be either a shorter push or only a little bit of jamming :P ), but wouldn't do any resetting on Arduino restart.

Lothean, 2018-06-13:
I'll do that when I can, never thought about it actually!

SammyIAm, 2018-07-04:
Fixed in 6ff396c13c09ab900eb6ed0cba1bd170b5e37285

closed 2018-07-04 by SammyIAm