samphonic/Moppy2 #15: Can't get 2.0 to work

OdinMinusNull, 2018-07-03:
Hi,
so I've set up my project with the old Moppy at first due to finding more tutorials about it, now I've switched to 2.0 as it seems it gives me a bit more control. Anyways, I can't seem to get it to work and I don't quite know why.
I can start the batch to get and interact with the UI, no problem. When I select my Arduinos port, the floppies also make a short sound when they reset their head, so they definitely react to it, or at least the Arduino is accepted by the software. I see that in the MoppyConfig there's supposed to be some startup sound? I don't get anything like that. 
When I load a midi file (either one of the samplesongs or a selfmade MIDI taht worked with the original Moppy), the floppies do nothing whatsoever. The output via Windows Midi Mapper works flawlessly, though.
I didn't change anything on my setup which worked with the original Moppy project; 5 floppies and an Arduino Uno. I've uploaded the latest Moppy.ino to the Arduino, and I still have the Timerone library installed.

Have I missed anything obvious? Any idea what the problem is? Thanks!

SammyIAm, 2018-07-03:
Apologies again about the 1.0 repository not adequately redirecting folks to this one-- I'll try to make that more obvious soon.

The startup sound you should be hearing (four rising notes on Drive 1) indicates that the Arduino code has been successfully uploaded and the drive is connected correctly; if you're not hearing that sound when the Arduino is reset (even when not plugged into the PC), then there's an issue either with the Arduino code or the wiring to the drives.

Since you had things working before, it's probably safe to assume your wiring is okay so it's likely to be an issue with the Arduino code.  Double check that the Arduino IDE was actually successful in uploading the code (sometimes it will print an error to the console without obviously failing and still always finishes with "Done uploading" which can be a bit misleading).

When you connect to your Arduino with the UI, does the Arduino show up on the right side of the screen listing its address and available sub-addresses?  (Lack of this would also indicate either an Ardunio-code issue or a communications issue between the PC and Arduino, but the lack of startup sound points to the former)

SammyIAm, 2018-07-03:
Alright, it's my fault.  The changing from device 2 to 1 was the tipoff.

I accidentally uploaded a release with some bad default configuration.  If you copy the contents of [MoppyConfig](https://github.com/SammyIAm/Moppy2/blob/master/Arduino/Moppy/MoppyConfig.h) and [MoppyCore](https://github.com/SammyIAm/Moppy2/blob/master/Arduino/Moppy/MoppyCore.cpp) from the links there, it should work fine out of the box.  (You can also look at [this commit](https://github.com/SammyIAm/Moppy2/commit/ec60aca5c8f60cc6a05c17512eb5472f55ea8f7d) and recreate those changes locally.

Thank you for the screenshot and the extra details-- I'll fix the release binary as soon as I'm home tonight.

Lothean, 2018-07-04:
I might be wrong, but if you rewired everything recently, you might have swapped the STEP and DIR pin on your floppy, since this noise is what I got when I wired them wrong.

OdinMinusNull, 2018-07-04:
Sorry for answering so late, I just tried it out and it worked perfectly with the old files.
So thanks for helping me out and bearing with my beginner's questions! :)

As for the round-robin question on the old rep, I'll just handle that when I generate my MIDI so it's already split up into different channels right from the start.

Either way, thank you so much for doing this awesome project, keep up the good work!

SammyIAm, 2018-07-04:
Hooray!  Glad it's working, and sorry again for the botched download.

You'll pretty much always get better results if you craft the MIDIs split into different channels, but a round-robin option is still fun for trying out MIDI files without the extra work.  To that end, I just committed a round-robin feature-- I haven't built it into a release yet so you'd have to build Moppy with NetBeans, but I'm hoping to have 2.0.1 out by tonight.  :)

Thanks for your patience, good luck with your floppies!

closed 2018-07-04 by SammyIAm