samphonic/Moppy2 #19: I cannot open the .bat file

mircea00, 2018-07-18:
I click to open the .bat file the command line shows up for half second then dissapears and nothing happens. I tried it on my desktop and it opened first try but i want to use my laptop for this project, what could it be?

SammyIAm, 2018-07-18:
First step is to make sure you've got the latest version of the Java JRE installed since that might cause it to fail.  If that doesn't fix it, you should be able to troubleshoot better by running the .bat file from the command line:
1. Open the folder containing the .bat file in Windows Explorer, and type `cmd` in the address bar to open a command prompt
1. Type `MoppyControlGUI.bat` and hit enter
1. You should be able to see the output of the .bat file and any errors that it runs into.

mircea00, 2018-07-19:
Thank you, the console closed so fast i couldn't see there were error messages, my Java was somehow older on my almost new laptop than on the desktop that i didn't update java for some years now. Now i need to get my hands on more floppyes and figure out how to wire this PSU so it can drive 16 floppyes, thank you for the work you have done.

SammyIAm, 2018-07-19:
Glad it's working!

closed 2018-07-19 by SammyIAm