samphonic/Moppy2 #20: Note Range Limitations

ApplesOfEpicness, 2018-07-21:
I'm wondering if it is possible to remove the note range limitations on either moppy 2 or moppy classic. I have some instruments that are capable of reaching notes outside the limits, but I'm not really sure how the serializers work so is it a software limit or a hardware limit? 

Thanks.

Lothean, 2018-07-21:
I already expanded the notePeriods array Arduino/Moppy/src/MoppyInstruments/MoppyInstrument.h which already increased the note range of the software.
You should check your mapper configuration for note limits or your instrument.
If the range is still not wide enough for you, expand the notePeriods, noteDoubleTicks and noteTicks arrays!

ApplesOfEpicness, 2018-07-26:
Forgive me for being so uneducated, but how exactly do i convert notes into microseconds?

dmadison, 2018-07-26:
Don't feel bad, I had the same question. See [this issue](https://github.com/SammyIAm/MoppyClassic/issues/161) on the "Classic" repo.

ApplesOfEpicness, 2018-07-26:
I'm not sure what i'm doing wrong. I've expanded the arrays from C0 all the way to C8.  However, my highest note is still cutting out right before C7?

Lothean, 2018-07-27:
Have you expanded all 3 arrays as they were supposed to ?

ApplesOfEpicness, 2018-07-27:
@Lothean 

This is what I did. I think it's right but i might be wrong.
[MoppyInstrument.txt](https://github.com/SammyIAm/Moppy2/files/2237434/MoppyInstrument.txt)

SammyIAm, 2018-07-27:
Looks good to me!  (That should work fine for you devices, hopefully)  If you want to do future users a favor and just go ahead and finish off the array, I'd be happy to accept a pull request :)

ApplesOfEpicness, 2018-07-28:
@SammyIAm 

But the notes are still cutting out before C7?

Lothean, 2018-07-28:
@ApplesOfEpicness 
Which instrument are you using ?

ApplesOfEpicness, 2018-07-28:
@Lothean

Software wise or hardeware wise? The software is set for normal floppies, but the instrument I’m physically using is a hard drive voice coil which has a nearly unlimited tone range.

Lothean, 2018-07-28:
Ah. I think you need to modify the file Arduino/Moppy/src/MoppyInstruments/FloppyDrives.cpp
Set the value of const byte MAX_FLOPPY_NOTE to 128 and it should work fine!

ApplesOfEpicness, 2018-07-30:
Huh, I was looking in the .h file instead of the .cpp. Yep, works pretty well. Some higher notes are out of tune so I’ll work on fixing that later.

Thanks for the help guys!

closed 2018-07-30 by ApplesOfEpicness

Lothean, 2018-08-04:
Concerning the notes that are a bit off, you should try setting the timer resolution to 20!