samphonic/Moppy2 #24: On board midi keyboard

DJthefirst, 2018-08-26:
I was wondering if it would be possible to add a virtual keyboard inside the program. I figure it would be easier to test the sound of different interments and be more interactive for displays. Also could it be possible to add a playlist feature like the previous moppy drive so that displays do not have to custom midi tracks to play multiple songs.

DJthefirst, 2018-08-26:
On another note I’m planning to build modular instrument units over the next two years ,as i am quite buisy, consisting of a xylophone with solenoids a drum module floppie drives stepper motors and a Tesla coil. Each module will have a 12v pass through power and a data line. Each box I am planning to make custom pcb with power supplies an arduino of course and the drivers for each insterment. Since I am more of a curcit designer than a coder I was wondering what your comments were on my project and how best to designe the data lines as I want them to be able to connect to any midi insterment I plan to build in the future.

Lothean, 2018-08-26:
You should use I2C which uses 2 data lines : SDA and SCL, and you can chain as much as you want within the max adress range.
You'll need a I2C master device that runs a code which handles the Serial to I2C things.
I already made my code for my project, but haven't heard back from Sammy about publishing it.
If you need it, I'll be happy to provide it to you!

DJthefirst, 2018-08-26:
So with i2c I would have my device (laptop) connected to the master arduino in the master/power model and then a data line connecting each of my other modules. Another question I have is 16 channels the limit or can I have 60 some let’s say. Because I have my drum and chime module that are each one Chanel and the Tesla coil is two but I’m planning 6-8 in the stepper motor and floppy drive modules alone. I could map a single channel that plays 8 notes at once but it’s less professional and my main goal is ease of setup.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Lothean, 2018-08-26:
Yes, that's the idea! Except it is 2 data lines ;)
When the track implementation will be finished, you will able to have 16+ channels

DJthefirst, 2018-08-26:
what do you think is the best cable types im planning xt60 for power but im not sure what to do for data

Lothean, 2018-08-27:
XT60 is a good choice for power
For data, you could just use 2 Dupont wires

DJthefirst, 2018-09-09:
I wouldn’t want too use DuPont wires they would be to breakable I feel. the plan is that I can quickly disassemble and reassemble them on the fly. I’ll find some connectors somewhere what about arduino? I’m making custom boards so I can integrate any of the types. I don’t know if going though the effort of an atmega 2560 and integrating all the mega components of if I should go for a nano with headers that can just plug in.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

DJthefirst, 2018-09-09:
The mega on one hand has numerous pins that can be used for channels. (Is their a sketch currently for the mega that uses all of the pins?)but is much more pricy and harder to upload sketches on a custom board. I could use the 328 ic chip which is the balance between pin count and cost but I still have to program it and add support components. The nano is cheep easily replaceable but does not have a desirable amount of pwm pins. I could chain two together but that adds complexity.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Lothean, 2018-09-10:
The nano uses the 328p and the Mega might have too much pin for the interrupt time.
And the Arduino's male and female headers are from the Dupont family, if I remember correctly

DJthefirst, 2018-09-10:
Ok the headers I’m fine with they are robust compared to the breadboard wires so then should I use 2 nanos In parallel in order to get the correct number of pins 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Lothean, 2018-09-10:
Could you explain a bit more what do you mean by "using 2 nanos in parallel to get 8 channels"?

DJthefirst, 2018-09-10:
The nano only has 10 usable digital io pins. So if I want to run 8 floppies off of it which use two pins each I would need two nanos. With steeper’s it gets worse if you use the l298n because it uses more then two. With a shift register I can see a nano easily working how ever 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

DJthefirst, 2018-09-10:
Also should I open a new issue for playlists and since this is solved other the Advice on my project keep it open?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Lothean, 2018-09-12:
@Sammy1Am : I am currently adapting Moppy2 for ESP8266/NodeMCU boards, if you are interested

SammyIAm, 2018-11-06:
> Also should I open a new issue for playlists and since this is solved other the Advice on my project keep it open?

If the original issue is solved, please close this ticket and open a new one for new issues/requests. :)

> ESP8266/NodeMCU boards

I just picked up a couple ESP8266s for some Christmas lighting so I might even be in a position to test those with my own floppies; plus I've been dying to try the UDP implementation to see how it works.