samphonic/Moppy2 #28: Assigning specific channels to different devices?

raidensnake, 2018-09-26:
Ok... I must admit not only is the new layout a bit confusing but I have several set up issues.

1. I have 2 Arduinos connected to one system.

2. The built in Arduino Leonardo is assigned to 6 floppy drives on comm port 5.

3. The external Arduino uno is assigned to 4 nema 17 l298n stepper motors on comm port 6.

4. Drums are played via stereo speakers.

The interface makes it hard to assign the channels to them. Any ideas how I can solve this?

SammyIAm, 2018-09-27:
Apologies for the new layout being a bit different-- there's a lot of compromise and I've tried to find a balance that works least-worst (and you said "a bit" confusing, I so haven't failed entirely 😉).  

We can definitely help you solve this; I don't have Moppy up in front of me now so I'm going from memory, but hopefully this will be enough to get you going (we'll start with just the Arduinos and get to the drums later):
1. In MoppyConfig.h (for the Arduino Code) for your Leonardo, you'll leave `DEVICE_ADDRESS` as `0x01`, and set `MAX_SUB_ADDRESS` to `6`.  This basically assigns that Arduino as "Device 1" and says it has 6 drives.  In MoppyConfig.h for the Uno, you'll change `DEVICE_ADDRESS` to `0x02` and `MAX_SUB_ADDRESS` to `4`.
1. In the GUI, when you've connected your Arduinos and checked the appropriate COM boxes, you should see both Arduinos show up in the Network area on the top right of the window with the appropriate number of sub addresses
1. You'll need two rows of Mapper in the mappers section at the bottom (click the "+" to add another row):
   * For the first row, your condition will be `c<=6` (which will evaluate to true for MIDI channels 1-6).  The device address for that row will be `1` (The Leonardo).  The subaddress will be `c` (which means the MIDI channel number will map directly to subaddress number), and you can just choose "Straight Through" for the note.
   * The second row will have address `c>=7 && c<=10`*, device address `2` (The Uno), subaddress `c-6` (MIDI channels 7-10 will map to subaddress 1-4), and once again "Straight Through" for the note

And then you're pretty much good to go for the notes.  Channels 1-6 will route to the Leonardo, 7-10 to the Uno.  The mappers are using Javascript, so you can do more complex mapping there if you need to.

* Channel 10 is MIDI's "traditional" drum channel.  There's no reason in theory you can't put drums on any channel, but if for some reason you can't, you'll need to work around that channel for the Arduinos playing notes.  So if you wanted 7-9 and 11, your condition would be `(c>=7 && c<=9) || c==11` and your subaddress could be something like `c==11?c-7:c-6`.
------

And now the drums.  Not sure what the speakers are connected to, but assuming it's either a software synth or a MIDI device, you'll need to enable the MIDI passthrough option in the synthesizer control on the top left, which will send **all** the MIDI notes being played to the MIDI OUT port of your choice (you can get a virtual MIDI cable if you're looking to route them back to the same PC with the Moppy Control software).  [VMPK](http://vmpk.sourceforge.net/) is a simple synthesizer program you can use to output the sound, however I don't know for sure if it will allow you to select *only* the drum track.

If the drums are being synthesized by a third Arduino, you'll want to give it e.g. `DEVICE_ADDRESS` `0x03`, and then route only the drums channel to that address.

Hopefully that gives you enough to go on.  The Mappers are understandably slightly more daunting in their script-based-ness than the previous UI, but the flexibility they allow fix a lot of peoples' issues without having a whole bunch of separate mapping systems.  Definitely let me know if you need more info though.

raidensnake, 2018-09-27:
Hey no problem thanks for clearing that up. In the previous version it was basically channel x you can enable or turn off. and set either the port or the midi out just for those channels. that's basically what threw me off. Also as I already know from before the floppy drives can do C1 to B5 where as the nema 17's seem to be set to C1 to F#5 (F Sharp) on the L298n for the stepper motors.

raidensnake, 2018-09-29:
Also here's a weird question. I have a spare 5th stepper motor and and unused pins on the Leonardo that aren't in use. Can it handle both types or do they need to be seperate?

SammyIAm, 2018-10-01:
Huh, well...

The short (easy) answer is: No, they need to be separate as each Arduino stack can only specify a single instrument type.

The long (difficult) answer is: There are no technical limitations that would prevent you from doing that.  😉 Your two best approaches (that I can think of off hand) are:

- Make your own custom instrument type that combines the code from FloppyDrives and L298N based on which subaddress is being called (maybe slightly easier, but definitely messier)
- Make a new MultiInstrument instrument that forwards system and devices messages sent to it to multiple instrument instances (instantiate both FloppyDrives and L298N).  There are a number of challenges here including the amount of memory available on the devices, and potential variable naming conflicts since this wasn't thought of at first; but the result would be more elegant, and reusable by any pair (or tuple!) of instruments.

Closing this issue since its original problem seems resolved, but feel free to reopen or open new ones if something else comes up.

closed 2018-10-01 by SammyIAm