samphonic/Moppy2 #29: Floppy drives current

antoniosavio96, 2018-09-28:
Hi man!
I really happy for your work here. I'm using Moppy since about a year, and the first version was pretty awesome!
Now I'm gonna upgrade my setup with other floppys and then I tryed Moppy 2.0.1 for see what's new.
But I have a lot of trubles, mainly because after trying 3 kinds of ATX power supplyes and 3 different floppy readers (and also after changing arduino mega to uno and trying different wirings), seems like the PSU can't provide the current for one floppy! 

Yes, it's ridicolous, because two months ago I played with 8 floppy drives and a 300W PSU (with 5A on the 5V rail, that's the only voltage I used all the time).

I don't know why without load the PSU has 5.05V, and if I connect the floppy drive and play a sample song sometimes it plays for some seconds (with the green light that changes brightness on every note, that's mean low power supplied) and sometimes it doesen't play at all, but the green light stays at low brightness and the voltage drops to 4.48V! 
I don't know what current is flowing, because I my ammeter won't work...

But the question is: Why with Moppy 1 all works nice and now seems like I need a 1200W PSU for hear something?

I tried all different stuff, wiring (like connect other GND pins), but as you read nothing seems work fine.... Maybe it's a my mistake? 
Greetings!

SammyIAm, 2018-09-28:
Hmm, yeah, something definitely doesn't sound right there.  I can nearly guarantee you that it's unrelated to Moppy 2, since none of the hardware or wiring has changed between the versions; and as a side note, the green light often changes brightness on drives even under normal operating conditions (so that not necessarily a sign of low power in itself).

Is there any way you could post a picture of your wiring for us to take a look at?  Maybe a wiring diagram?  Where is your Arduino getting its power from?

Some thoughts in the meantime:  If the drive tried to pull more power than the PS could supply, the PS would shut off (short circuit protection); some will even shut off for over or under voltage conditions.  Since it doesn't sound like your PS is shutting down, it seems more likely the issue is somewhere between the PS and the drive, something with the drive itself, or an issue with the Arduino/drive that's unrelated to power.

4.5v on a 5v rail _might_ indicate something drawing a lot of current, but is potentially close enough to spec that things would still work (ATX specifies +/-5%, and you're at -10%, but floppy drives aren't super complex, so there's a decent change they'd operate fine)

antoniosavio96, 2018-09-28:
Hey man! 
**You're right.. those problems are now solved...** Maybe the main problem was the USB cable for the arduino, and also there was a problem of ground in the ATX plug, that maybe will cause those problems with strange current....

After this, because the ATX is pretty old (made in 2002), I plugged the floppy plug in the wrong way and the cables litterally burned my fingers. Old school short-circuit protection? Maybe...

Now all seems work good. But I really miss some feature from Moppy classic, like the tempo change and the single channel selector, that was on the right... I'm not so friendly with the new interface, I didn't know anything about the new features and an easy wiki about them will be really appreciated!

After this, if I plug the mega and select the COM on Moppy2, the sub-addresses are from 1 to 8.... Why? 
Mega has a lot of outputs... Why is recognized like a normal arduino uno?

If you can kindly reply to my questions, this trouble will be closed. 
This is not a threat lol
And thanks again for your brilliant works here!

SammyIAm, 2018-10-01:
Glad you got it working!  The tempo can still be changed using the MIDI file, but there's not a slider for it (feel free to open a new issue requesting that if you'd like).  As for channel selection, you can do anything you could with the previous version and more, but you'll need to know a teensy bit of Javascript.  I just explained some channel mapping in [this issue](https://github.com/SammyIAm/Moppy2/issues/28) that might help put things in perspective.  If you want just a single channel, can set the Condition script to e.g. `c==2` if you wanted only channel 2.

There's no way for the Arduino to detect how many drives are hooked up to it, so out of the box the code just assumes 8 drives.  You'll change the MoppyConfig.h file to specify the min and max drive subaddresses (again see the linked issue above for more details).  The code only currently supports 9 drives though (the max supported by the Uno), so you'll need to make some small changes to run more drives on a Mega.

SammyIAm, 2018-10-01:
You mentioned freezes; if they're reproducible, you can open new issues for those and we can try to resolve them.

closed 2018-10-01 by SammyIAm