samphonic/Moppy2 #31: Multiple arduino controllers

sniff122, 2018-10-24:
Is using multiple arduinio controllers possible, i have 9 drives and 2 unos, if i set the device address of one to 0x02, will it work because it is not working for me

SammyIAm, 2018-10-25:
You're on the right track, you want to set the device address of one of the Unos to 0x02.  When they're both plugged in and their COM ports selected in the center of the screen, you should see them both in the upper right.

However, you'll also need to add an additional Mapper row at the bottom to determine which notes will be sent to device 2 instead of device 1 (the default mapper only sends to device 1).  You'll also want to set conditions for each Mapper if you want to split some channels between each of the controllers.

If you're still a bit stuck on that, a screenshot of your current MoppyGUI configuration would make it a lot easier to help you out.

_A quick side note: I'm not trying to discourage you from using two Unos because it's definitely supported and lets you do some cool stuff, but it's also possible to add a 9th drive to a single Uno and up the max subaddress by one.  Maybe try with two boards first at least, since that leaves room for expansion anyway._

sniff122, 2018-10-25:
I have figured it out now, thanks though

closed 2018-10-25 by sniff122