samphonic/Moppy2 #32: Connect more than one driver ?

gustiome, 2018-10-25:
Hello !
I'm stuck in the same situation since a few week ago.
I connect all the driver to the Arduino Card like it says in the project (first driver : on 2&3 pin, seconde driver: on 4&5 etc..) and there all wired with an ATX Alimentation..

With the Java interface, are we must connect each driver on different output or a single output (com3 for exemple ARDUINO Card) ?

Two drivers are connect to my project but only once work..

Thanks for helping !

Gus

SammyIAm, 2018-10-26:
Hmm, which of the sample songs are you playing?  Most of the included ones should work on at least two drives.

If you switch the two drives (drive 2&3 => 4&5 and drive 4&5 => 2&3), does the same drive work, or does only the drive on 2&3 work?

If it's still not working, could you send a screenshot of the program on your PC?

gustiome, 2018-10-26:
I use some midi files with 3 tracks, I don't think the problem comes from here.

[image]

I switch like you say but  only the drive 2&3 works !

This is the moppy software that I use for.

[image]

thanks for giving your time for me, it's very appriciate !
(sorry for my english, I come from France)

Gus

gustiome, 2018-10-26:
It's finnally work with two drives !!
I click on the second moppy serial on the moppy software, before it doesn't work but today it works !
I don't know why now and not before but it works !

Thank you so much for helping ! I'll telling you if there's no probleme with more drives ;)

Gus

SammyIAm, 2018-10-26:
I'm glad you got it working!

I do see though that you're using the older version of the Moppy software.  As long as it's working for you there's no need to upgrade necessarily, but [the newer version](https://github.com/SammyIAm/Moppy2/releases) has some additional features that might be nice (though it sounds like it can also be a bit more confusing as well)

Definitely feel free to reopen this or open a new issue if you run into anymore issues.  Have fun!

closed 2018-10-26 by SammyIAm

Lothean, 2018-10-27:
An additional note on that, if I'm allowed, for fellow French Moppy2 users :
Nous avons récemment mis en place un Discord concernant l'assistance en français pour Moppy2, puisque la demande est importante (je reçoit 15 demandes par mois environ). Si jamais tu viens à avoir de nouveaux problèmes, n'hésite pas à m'envoyer un mail pour que je te donne le lien du Discord ;)