samphonic/Moppy2 #34: Motor Shield

istwan100, 2018-10-30:
Hello,

is it possible to use the original Arduino motorshield instead of the L298N?
I'm a little bit confused about the schematic of the motorshield, because its not wired like the inputs in the L298N.h

istwan100, 2018-11-07:
I tried to configure the outputs in L298N for the Motorshield, but I couldn't got it to work after trying multiple port configurations..
The Motorshield worked partially, the Motor tried to move but he didn't.
I have no Idea, what I can try now. First I tried to wire it like a L298N, it does not work, than I tried to enable the Outputs on the Motorshield. After this, the motor tried to turn but it doesn't worked.

In the schematic of the Motorshield there is a 4077D AND gate between the DIR A and BRAKE A Pin (which should go to the IN 2 of the L298P I think) goint to the IN 2 and I don't understand, why the gate is there. 
Schematic: [https://www.arduino.cc/en/uploads/Main/arduino_MotorShield_Rev3-schematic.pdf](url)

Has anyone an idea, why it isn't working?
Thanks a lot for help!

Lothean, 2018-11-09:
Hi!
Sorry for the late response, I had very little time to look into your problem, but since it's almost 5AM and I'm procrastinating more than ever, here's your solution :
I had a look at the schematic of your shield, and it's L298N based, so it's a good thing
Then, I tried to figure out why the wiring wasn't the same and I found out.
On the classic L298N breakout boards, the inputs for controlling both bridges are called IN 1 to 4, which is what I'm reffering to in the pinout.
On the motor shield, they are called differently, and the names are as follows : 
IN1 = DIR A
IN2 = DIR A#
IN3 = DIR B
IN4 = DIR B#
Also consider that DIR A#/DIR B# is inverted by the shield, and thus allows us to control the stepper with only two wires (I doubt a lot on that though, schematic is confusing)
You might also need to connect PWMA (D3) and PWMB (D11) to ground (If I remember correctly), which is normally done with a jumper on the breakout boards.
To make things simpler for you, I made an alteration to my L298N code which allows you to control a stepper motor with the motor shield:
L298N.h : https://pastebin.com/B8DpEkXv
L298N.cpp : https://pastebin.com/qCvTJZWJ
(Don't pay attention to the comments or to residual code, it's a quick bodge, just plug the shield and that's it)
It compiles fine, but I have no idea if my theory is correct, so be sure to let me know ;)

istwan100, 2018-11-09:
Thanks a lot.
I think on tuestday I'll have enough time to try it^^

Lothean, 2018-11-09:
The AND gate is very confusing though, I doubt it is compatible

istwan100, 2018-11-13:
It works!
Thanks for your help!

The only Thing I had to change was the PWM pin. It had to be connected to 5V. I fixed it with setting the pin instead to LOW on HIGH. Everything else is working.

The next Problem is, that the L298P and the Motor are getting very very hot (I know. Do you know, what I could do, that they do not overheat?

Lothean, 2018-11-13:
That's because the motor you use draws too much current. Please see what I explained in #36

istwan100, 2018-11-14:
If I use the DRV8825, would the stepper be quieter?

Lothean, 2018-11-14:
No! It depends on the supply voltage (higher voltage would mean that the motor would be more reactive) and of your current limit.
I stay just a little before the rated current of my stepper but with a high supply voltage so it's reactive in high frequencies and does not jam in low frequencies.
Remember that the supply voltage isn't proportional to the current that draws the motor when you use a DRV8825 since it's current limited. So you have the benefits of high voltage stepper driving without the risk of overcurrent :p

istwan100, 2019-02-06:
First of all, I'm sorry for not writing this long time, but I had other stuff to do :/

I've tried it with different scanners and with 2 it worked, but I do not understand how to configure moppy, that it can use x Steps until the Scanner ends. Because at this moment it only uses a little part of the scanner an not the whole. The second problem I had is that if I let the stepper go to its home postition, I think Moppy does not know, where it is. Also if the scan unit was at the end of the scanner the motor keept moving. 

The only things I've changed was to higher the amount of drives, that was connected to the Arduino, witch is one. Also I tried to higher the MAX_POSITION for the scanner in the FloppyDrives.cpp. 
But it does not work. If it helps you, I can upload a Video from the Scanner or photos / sketches from my wiring. Or can you upload me a sketch of wiring a stepper to the DRV8825?
I used this: [http://www.hobbytronics.co.uk/image/data/pololu/drv8825-stepper-motor-driver-4.jpg](url)

SammyIAm, 2019-02-21:
Higher `MAX_POSITION` should give you what you're looking for.  You might have to make it a *lot* higher though depending on the size of the steps.  Don't go over the max size for integers, of course, but you might even try just putting the max value in there to make sure it's working (and get ready to disconnect it when it hits the end of the scanner.  Make sure you've got the right index for the scanner (maybe try disconnecting the floppy drives and increasing **all** the `MAX_POSITION` values to make sure).

Unfortunately there's not any current support for Moppy learning where the head is if it's reset via other means.  It's possible that if you had a sensor or pin from the controller/scanner that could tell the Arduino when the head has reached the end, that you could set up an interrupt to reset the `currentPosition` value for the scanner when it got there.

Sammy1Am, 2020-08-12:
Closing for lack of activity.

closed 2020-08-12 by Sammy1Am