samphonic/Moppy2 #35: multi floppy connect with arduino

Ehsan-Ebrahimi, 2018-11-05:
Hello everybody,
I have a multi sound MIDI file, via Netbeans I want to play, But only one floppy disk works. 
Is there anyone who knows what should I do, setting in Netbeans must be changed or there is another alternative?
Thanks

SammyIAm, 2018-11-06:
Hi there!  Not a NetBeans setting, but possibly a Moppy setting.

Could you provide a screenshot of your current configuration?  That'll help see what you've got so far.

Also, do you know if the MIDI file you're trying to use has multiple *channels* or just multiple *tracks*?  Moppy will only sort-out notes by channel.

Ehsan-Ebrahimi, 2018-11-07:
many thanks for your response.
I managed to do it und now its works
but it would be nice if you can check everything

one more question, did you connect HDD with arduino?
Do you know how it works?

Many Thanks

Sammy1Am, 2020-08-12:
Glad it's working.  I don't personally have any HDD connection experience.

Closing for lack of activity.

closed 2020-08-12 by Sammy1Am