samphonic/Moppy2 #37: Floppy drive too silent..

Machinegamer101, 2018-11-12:
Hello, i have a box of 25 floppy disk drives.. but i only finded 4 that actually do a decent sound.
i mean, the others barely emit a sound at all
anyone knows how to fix this?
(All the floppy disk drives are working, the motors work but they barely make any sound at all)

Machinegamer101, 2018-11-12:
i also removed the top cover to all the floppy drives.

SammyIAm, 2018-11-12:
> anyone knows how to fix this?

Well it's not a "fix" per se, since quiet/silent drive operation is generally considered a *feature* (since *apparently* some people don't like their computers as noisy as possible ;P ).  But I get what you meant.

Some various thoughts and findings about amplifying your drive sounds (roughly in order of least to most work):

- I found that my drives are louder without a floppy disk in them (though it's also worth trying to put a disk in to see if yours are louder the other way around).
- Make sure the drives are mounted / sat on a hard smooth surface.
- Try lifting the drive a bit or angling it to allow sound out the other side as well.
- My drives seem to have particular note ranges that they're particularly loud in (usually lower notes).  You can try adjusting the music to see if you can find an octave that your drives do better in.
- [Attaching SOLO cups like this guy](https://www.youtube.com/watch?v=j6OTPmNWydc) actually seems to do a pretty good job of amplifying the sound.
- Attaching weights to the read head might make the drives louder as they struggle more, but this definitely increases the risk that you might damage the drive.
- Similarly, cleaning off the drive screw with rubbing alcohol to remove any lubricant should in theory make the drives run *worse* (i.e. louder), but maybe isn't the best for them.
- In theory you could passively amplify the sound in the same way stringed instruments do (e.g. acoustic guitars would be super quiet without their resonating body).  I haven't seen this one actually attempted, and there's some interesting research to be done with regards to size and shape (i.e. a closed tube might work for one specific note, but not the rest).  Basically you'd mount the drives over some sort of cavity to amplify the sound.
- Moving the microphone closer to the drives will make them appear louder in the recording (this one's sort of obvious, sorry, but even the louder drives aren't *that* loud)
- I did see someone else manage to use electric pickups to pipe the drive vibrations directly into a soundboard.  The kind you clip onto a string bass would probably be idea, but even electric guitar pickups might work since the drive body is metal and the motor is pumping out a little bit of a magnetic field.  If you go this route, experiment with different attachment points on the drive.

If I think of anything else today I'll add it to the list, but that should hopefully give you something to go on.  Experimenting to figure out what works best for your setup is, unfortunately, probably the way to go.

Machinegamer101, 2018-11-12:
Thank you so much! i "fixed" a drive by removing the lubricants on it, i will keep doing this for the other drives. 
i usually dont use the disks since i dont have them, but opening the "disk bay" of the floppy drives by grabbing the disk opener and rotating it clockwise.
[image]
[image]

closed 2018-11-12 by Machinegamer101