samphonic/Moppy2 #42: I don't understand the steps

owenr930, 2019-01-06:
I am doing this as a project due Friday and I can't turn it in late because the quarter is ending. I can't seem to get any of this to work. SOMEONE PLEASE HELP!

SammyIAm, 2019-01-06:
Hello there, I'm certainly happy to help out however I can, but first, let's figure out what we've got going on:

- What have you done successfully so far?
- What are you stuck on / isn't working?
- What error message / behavior are you seeing?

owenr930, 2019-01-06:
I've gotten somewhere since making that comment but I'm still having
problems. So I followed the tutorial in the description (step by step
projects). So far I uploaded the arduino code successfully, I have wired up
my drive to the arduino and power supply, and I have made it all the way up
to step 18. When I press run it gives me some sort of error or bug and the
window to play music doesn't open up (I know next to nothing about java or
netbeans). This is the error I'm getting when I press run.

ant -f C:\\Users\\owenr\\OneDrive\\Documents\\NetBeansProjects\\MoppyDesk -
Dnb.internal.action.name=run run
init:
Deleting:
C:\Users\owenr\OneDrive\Documents\NetBeansProjects\MoppyDesk\build\built-jar.properties
deps-jar:
Updating property file:
C:\Users\owenr\OneDrive\Documents\NetBeansProjects\MoppyDesk\build\built-jar.properties
compile:
run:
Jan 06, 2019 5:11:59 PM java.util.prefs.WindowsPreferences <init>
WARNING: Could not open/create prefs root node Software\JavaSoft\Prefs at
root 0x80000002. Windows RegCreateKeyEx(...) returned error code 5.
java.lang.UnsatisfiedLinkError: C:\Program
Files\Java\jdk1.8.0_05\jre\bin\rxtxSerial.dll: Can't load IA 32-bit .dll on
a AMD 64-bit platform thrown while loading gnu.io.RXTXCommDriver
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError:
C:\Program Files\Java\jdk1.8.0_05\jre\bin\rxtxSerial.dll: Can't load IA
32-bit .dll on a AMD 64-bit platform
at java.lang.ClassLoader$NativeLibrary.load(Native Method)
at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1929)
at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1835)
at java.lang.Runtime.loadLibrary0(Runtime.java:870)
at java.lang.System.loadLibrary(System.java:1119)
at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:123)
at moppydesk.MoppyMainWindow.updateComSelectionMenu(MoppyMainWindow.java:55)
at moppydesk.MoppyMainWindow.<init>(MoppyMainWindow.java:43)
at moppydesk.MoppyUI.startup(MoppyUI.java:19)
at org.jdesktop.application.Application$1.run(Application.java:171)
at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
at java.awt.EventQueue.access$400(EventQueue.java:97)
at java.awt.EventQueue$3.run(EventQueue.java:697)
at java.awt.EventQueue$3.run(EventQueue.java:691)
at java.security.AccessController.doPrivileged(Native Method)
at
java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
at
java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
at
java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
at
java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

SammyIAm, 2019-01-06:
The error you're running into is related to the serial drivers (looks like it's trying to load the 32-bit version on a 64-bit system).  

However it also appears you're attempting to use the older version of Moppy.  Go ahead and grab the latest version on the [releases page](https://github.com/SammyIAm/Moppy2/releases).  You'll need to use both the newer Arduino code and the newer Java code (should launch just using the .bat file, no need for NetBeans).

The newer version has more stable serial libraries, so that should resolve some of the problems.

owenr930, 2019-04-04:
Sorry it's been so long since my last message. Since then I have gotten it to work with 8 drives. I was curious about 2 things. First, is it possible to use the same code to make an old scanner play music? Second, I've been trying to use the trial version of FL studio to edit sound files so they are in the range of the drives but when I load in the file it plays them all on the smart drive. Do you know how to fix this?

SammyIAm, 2019-04-05:
Glad you got it working!

First, more or less the answer is yes.  The code generates timed pulses at the desired note frequency to drive the floppy drive, so the same pulses can be used to drive any sort of stepper motor (such as the ones in scanners).  However, I'm not personally familiar with how the hardware side of things works for a scanner setup (though there have been others here that have got it working, so search around a bit.

Second, what do you mean, "it plays them all on the smart drive"?

owenr930, 2019-04-05:
Oops, I meant same drive

SammyIAm, 2019-04-05:
Which drive a particular note is played on is determined by the MIDI Channel that that note is assigned to.  I'm not familar with FL Studio, but you'll need to figure out how to get each track assigned to a separate channel.

owenr930, 2019-04-06:
Ok, I'll check it out.

owenr930, 2019-04-06:
Also, I may try a scanner with the same system. I attached a file that show
the wires of a stepper motor similar to the scanner ones. Which of the
wires in that picture would be the wires I need and how would I hook them
up?

SammyIAm, 2019-04-06:
I don't see an attachment, I'm not sure that worked.  Regardless, I wouldn't know which ones need to be hooked up; you'd have to look up the specs for that particular motor.

However, unless the stepper motor has some integrated circuitry, you're most likely going to need a stepper motor driver.  The L298N does actually have Moppy support already thanks to the help of @Lothean , so that might be an easy one to start with if you don't already have a driver you're trying to use.

owenr930, 2019-04-10:
So if I get one of these drivers would you be able to help me set it up? I
got ahold of an old scanner with a stepper motor.

SammyIAm, 2019-04-10:
I'm always happy to answer any questions you end up having, but I've never set up a stepper driver so you'll probably find more accurate help / guides on Google.

owenr930, 2019-04-10:
> Hi I connected the wires of the stepper motor to the Easy Driver by Brian
> Schmalzhaus(1 coil to the a pins and the other coil to the b pins). To
> power the stepper motor and the driver I currently use a lab power supply(I
> can change the volt etc). Be sure what power input your stepper needs. I am
> planning to connect a pc psu rather than the lab psu to the GND and M+ pins
> to power the stepper and driver. The driver can be connected as a floppy
> drive to the arduino and the GND pin to GND on the arduino. In my code I
> used 2 more pins for the speed control. Here is a simple pin table:
> Arduino pin Driver pin
> 2 Step
> 3 Dir
> 4 MS1
> 5 MS2
> GND GND next to Step
>
> I need to make the code a little clearer so i'll post it when it is
> (probably tomorrow). It still has some problems like:
>
> How do I modify the reset function to only move the amount, it moved
> forward, backwards?
>
> I hope you could understand how i got it to work and I could help you.

Thanks for getting back to me! I have a few questions. First, do you reckon
I'd be able to use an A4988 driver instead of the easy driver? Also are the
two extra pins necessary for the motor to work or can I stick to just the
direction and step pins? I have an Arduino uno running 8 drives at the
moment, I plan on getting two scanners, and I want to take off the least
amount of drives as possible. If I had 4 wires for each scanner I would be
losing half of my drives.

tobiaslp12, 2019-04-10:
I think you can use the A4988 and according to [https://www.schmalzhaus.com/EasyDriver/Examples/EasyDriverExamples.html](url) it is controlled the same way. I would use a different Arduino for different instruments because a scanner can play higher notes but the floppys can not and in the floppy code there is no speration. Every floppy or scanner on the same Arduino would have the same max note. The 2 extra wires are not necessary. The driver would do an eights step by default but I don't know if it is the same on the A4988 (or BigEasyDriver). 
So yes you can just wire the A4988 up as a floppy drive but:
1. The floppy drives may try to play notes that they are not capable to play properly if you set the `MAX_FLOPPY_NOTE = 71;` to `MAX_FLOPPY_NOTE = 107;` in `src/MoppyInstruments/FloppyDrives.cpp` which is necessary for the scanner to play high notes.
2. You may want to change the values in `FloppyDrives::MAX_POSITION[]` for each scanner to a higher value like: if scanner is "drive" 8 then change it to:  `{158,158,158,158,158,158,158,5000,158,158}`. 

(I hope this is correct if not please correct me)

owenr930, 2019-04-11:
Ok, thanks. As far as notes go, I have recently learned how to edit midi
files on FL studio's trial version. So now I can control which channel(s)
will have the high notes on them. I can also move notes down a few octaves
for the drives. In order to let the scanner play higher notes I can select
the option to play though notes that are out of range for the floppy
drives. This means that I shouldn't need two arduinos as long as I edit all
notes in range for the drives. I was also thinking about taking off drives
to make room for the scanners but instead I think I'm just going to order
an Arduino mega so I have more pins.

owenr930, 2019-04-12:
I just hooked up everything to get my scanner to play music but it is
unresponsive. I figured out which wire was which on the stepper motor by
using a multimeter on ohms. I then hooked up the 4 wires to the driver
(each two wires showing a resistance on 1a/1b and 2a/2b). I plugged in my
24 volt 2a power supply to the vmot and ground on the driver. Then I
plugged in my step and direction pins for channel 1 into the driver. Then I
connected vdd and ground to the 5v and ground of my Arduino mega. When I
try to play something the floppy drives are playing but the scanner won't
make any noise at all. Is there something I'm doing wrong?

owenr930, 2019-04-12:
Nevermind that, I just got it to work. But how do you change the max
possition of the scanner? You mentioned it in your last email but I don't
know where/what that is

owenr930, 2019-04-14:
How so you change the distance the scanner head travels?

owenr930, 2019-04-16:
So I've hooked up a scanner but the "scanner head" (the thing that moves
back and forth) only moves a little bit which I believe is because the
program still thinks its a floppy drive. Do you know how to change the
distance it can move?

SammyIAm, 2019-04-16:
A handy tip for GitHub is that you can [search within a repository](https://github.com/SammyIAm/Moppy2/search?q=MAX_POSITION&unscoped_q=MAX_POSITION) for a particular term (like `MAX_POSITION`) if someone mentions something that you can't seem to find.

owenr930, 2019-04-16:
I found that in the code but I don't know what max possition means and how raising or lowering any of those values would effect any of the drives/scanner.

SammyIAm, 2019-04-16:
"Maximum position" in this case means the maximum number of steps forward the drive head will take before it reverses direction (as the comments in the code explain, the values stored in the `MAX_POSITION` array are actually double the number of steps).

Increasing this value for the appropriate "drive" (or scanner in your case) will tell Moppy to take more steps forward before reversing the direction of the steps.  Because we're using an unsigned int, on an Arduino Uno the maximum possible value is 65,535 (though 65,534 probably makes more sense because it's divisible by 2).

owenr930, 2019-04-16:
So if I have my scanner's motor driver hooked up to pins 2 and 3 and I raise the first value it should go farther before changing direction?

SammyIAm, 2019-04-24:
If by "first value" you mean the value at index 1, yes.  Note that as it says in the comments, the value at index 0 (the zeroth value, I guess) has a special purpose.

owenr930, 2019-04-24:
Oh, now that you say that I feel kinda stupid. I always forget how arduino
starts counting at 0 not 1. That explains why there was 9 values instead of
8 and nothing happened when I changed index 0. So now that I know this was
my problem, if I want to add two more devices will I need to add more
indexes on to the list?

SammyIAm, 2019-04-25:
There should be 10 values 😉 

If you wanted to add drives 10 and 11, you would need to add two more values to that array, but also to several other arrays in that same file, as well as changing the max sub address value, and adding some code blocks in the `tick()` function.

owenr930, 2019-04-25:
Ok, I'm going to add one more drive now that I realized that there is
already 9 in the program. So to add one more I basically need to add 1 more
value everywhere I see a list of numbers, I need to set the last drive
value to 10, I need to add pin modes for the extra 2 pins I need, and I
need to add an extra block of code in the tick function. Am I missing
anything before I go and make the changes?

SammyIAm, 2019-04-25:
You'll also need to adjust the minimum and maximum sub address values for that device so that it's listening for events sent to higher-numbered drives.  I _think_ that's probably everything, but I might be missing something; you may have to do some troubleshooting to get it working.

owenr930, 2019-04-25:
What are the sub address values, what are they for, and what will they look
like in the code. I'm rather inexperienced.

DinukaShehan, 2019-09-30:
> Hello there, I'm certainly happy to help out however I can, but first, let's figure out what we've got going on:
> 
> * What have you done successfully so far?
> * What are you stuck on / isn't working?
> * What error message / behavior are you seeing?

please make video how to install all the stuff? not understand properly

SammyIAm, 2019-09-30:
@DinukaShehan If you're having issues, please:

1. Create a [**new** issue](https://github.com/SammyIAm/Moppy2/issues/new), rather than piggy-backing on an existing one (unless your problems are related to the issue that's already been created)
2. In the new issue, please answer the following questions:
  2.1 What have you done successfully so far?
  2.2 What are you stuck on / isn't working?
  2.3 What error message / behavior are you seeing?

Sammy1Am, 2020-08-12:
Closing for lack of activity.  [A tutorial video](https://www.youtube.com/watch?v=D2iswCnV9ec) is available now.

closed 2020-08-12 by Sammy1Am