samphonic/Moppy2 #43: Problem with >11 floppies + Question

THLCX, 2019-01-06:
Hello

There is a problem with the file FloppyDrives.cpp :  
Only 11 floppies are supported, 16 are not because of implementation holes.

I fixed everything :  
[ThomasCyrix.com/files/FloppyDrives.cpp](url)

Also, I can't find a way to simply select or unselected which channel will be active when playing the music.
It was a good feature in the 1st version of Moppy, is an update to add some easy and fast interface about this planned ?

Thanks for your work

C288, 2019-01-19:
I have an arduino mega 2560. I have 16 drives connected but can't get all of them to work. How would I do this?

SammyIAm, 2019-01-22:
Closing due to lack of response, feel free to reopen if you've still got questions.

closed 2019-01-22 by SammyIAm