samphonic/Moppy2 #44: HDD arm as kick drum?

legop3, 2019-01-08:
hey, new to github and moppy. is there a pre-made bit of code that I can use to assign MIDI drums to HDD arms? arent MIDI drums on their own channel? thanks.

SammyIAm, 2019-01-08:
Off hand I don't have any code to share, but if you look at some of the forks of the repo (or maybe the old repo), there might be some things there.  However, I can provide a bit of info:

- MIDI drums are an instrument (or "voice") like "piano" or "violin", except that instead of MIDI note 69 being a pitch like A, it's a Cabasa (whatever that is 😉).  _Traditionally_, MIDI drums occupy channel 10, but there's no particular enforcement of that in most cases (so you can turn any MIDI channel into drums by assigning it to the correct voice.
- If you're going to use MIDI drum music for a HDD arm, you'll need to decide if you want to convert *all* events on the drum channel to HDD movement (e.g. it doesn't matter if it's a high hat or a bass drum "note", the HDD will move) or if you want to assign a particular note to the HDD (e.g. it only plays snare).  You can accomplish this in the Moppy mapping section by setting a condition based on `n`.  So for example `c==10 && n==43` would route all Floor Tom notes from channel 10 to whatever device address you assign your HDD in the Arduino code.
- The current Arduino code for playing notes turns the pin connected to the floppy drive on and off rapidly as long as the note is "on".  To switch to using a HDD like a drum, you'd just turn the pin on once when the note turns on, and turn the pin off when the note turns off (or possibly after a pre-defined delay).

I don't have anything more specific to provide now, though I can tell you that if you hook a HDD arm up to the existing Arduino code, the arm should play notes like the floppy drive would (though probably very quietly), and you'll be pretty close to using it as a drum.

Do you have a circuit you plan to use to control the HDD arm?

Sammy1Am, 2020-08-12:
Closing for lack of activity.

closed 2020-08-12 by Sammy1Am