samphonic/Moppy2 #45: Arduino Mega 2560 with 16 Drives

C288, 2019-01-22:
I have an arduino mega 2560. I have 16 drives connected but can't get all of them to work, only 8 of them work. How can I get this to work?

SammyIAm, 2019-01-22:
Just a few changes:

1. Update `FloppyDrives.cpp` to contain variables and code for all 16 drives.  You can see an example of how someone has done this in [this issue](https://github.com/SammyIAm/Moppy2/issues/43).
2. Update `MoppyConfig.h` so that the `MAX_SUB_ADDRESS` is appropriate (probably 16 if you're using 1 through 16)
3. If you have a MIDI file with 16 channels already, you're good to go.  If not, you'll need to update the mapping in the GUI to duplicate/route lower channels to the new channels you've added.

C288, 2019-01-23:
Thanks. Just one more question, is there a way to use the round robin feature with 16 drives? So if I'm using a midi with 8 channels they will all still play.

SammyIAm, 2019-01-23:
In theory yes, you'd use the example round robin script, but with `nextRoundRobinSubAddress(16)` instead of 8.  **However**, while checking the code, I think I found a hard-coded `8` on [this line](https://github.com/SammyIAm/Moppy2/blob/698c4a2863d62ef8bdf9c3d8ab7eaf4c0a7f543d/Java/MoppyLib/src/main/java/com/moppy/core/events/mapper/MIDIScriptMapper.java#L159) that you'd want to change to `numberOfChannels` instead.

C288, 2019-01-23:
Thanks That worked perfectly.

SammyIAm, 2019-02-21:
Awesome!

closed 2019-02-21 by SammyIAm