samphonic/Moppy2 #47: Speculative questions

NNNenov, 2019-03-05:
Hi there, I'm planning to to put together a few moppys when I get back home from holiday, I just read through the instructions and it seems quite straightforward, despite me being very Rusty with Arduino and not good at electronics. I have some questions I wanted to ask before I get started:
1. i) Does every floppy need to have a ground pin connected to the Arduino's ground pin? 
 ii) You mentioned every odd pin is a ground pin on the floppy, so any one of them would be ok to use?
2. If we are only connecting step and direction pins, how is the head end/start limit considered? How does the Arduino know where the head position is? I haven't looked at the code and I don't know anything about floppy drives, so I don't know if the drive itself automatically will prevent the head from trying to go beyond the limit, is this something I wouldn't need to worry about? How does that work? What happens when the head reaches start/end of the track?

Thanks in advance! I'm excited to try this project!
Nick

pyraei, 2019-03-05:
I didn't do any contributions myself yet (but i really want to if i find some time) but since i played around a bit with moppy already i can answer your questions.

1. i) yes
   ii) also yes

2. Moppy will force all the way to starting position during initialization, it knows how many steps from there to reach the end. It always remembers the position and will reverse direction to not hit either end or starting position.

NNNenov, 2019-03-06:
Thank you pyraei, most helpful.
So it will send movement signal enough times until it is with no doubt at the start position, this is not harmful for the motor/hardware?

I was just reading the guide again and realised I don't really understand step 1.
"Connect the appropriate even-numbered drive select pin to its corresponding odd-numbered ground pin." 
How do I determine the appropriate drive select pin? Is it determined based on which Arduino output it is connected to? From what I have read so far there is A and B. And what if you have more than two drives?

Thanks again,
Nick

pyraei, 2019-03-06:
yes, and pretty sure it's not harmful, yes.

just as before, it doesn't even matter which ground pin you use.
Further it doesn't matter if A or B just make sure either 12 or 14 are connected to a ground pin, this is needed to make the hardware work.
This is really just from the way the floppy drives were meant to be used, old PCs supported 2 drives and the cables would have 3 ends, 1 going into the Mainboard, one to drive A and one to drive B (that one would have some cables twisted to change the drive select pin, easy to see when you google for it)

Moppy only cares for the arduino output pins being used (2&3 first floppy drive, 4&5 second floppy drive, etc...)

SammyIAm, 2019-03-06:
Just to add a little to @pyraei 's excellent info:
1. i) Still yes.
ii) *Almost certainly* yes. 😉 I don't remember coming across any floppy drives where the ground pins weren't all connected on the floppy-drive side, but it's always possible that some might be separate or not connected for some reason.  I tend to try to use one of the ground pins opposite the pins that I'm using for Step or Direction control.
2. The floppy drive interface doesn't provide any information about the head's current position so, as @pyraei said, the "correct" way to reset the head back to the starting position is just to step it that direction as many times as there are tracks.  Some drives have a small sensor at that position so that the stepper motor won't try to move the head any further back, others are just built to tolerate it.

Sometimes (especially for higher frequency notes) the actual drive position and the position Moppy is tracking will diverge though.  If this results in the head going too far *back* it's basically like resetting the drive and things sync back up.  If the head goes too far *forward*, you get a nasty clicking/grinding sound.  The drives appear to have been designed to accept this sort of treatment, so it doesn't really hurt the drive, it just doesn't sound nice.  If it's happening a lot, the notes you're playing on that drive might be too high.

Regarding A vs B (with some historical background you probably don't care about but there's a tl;dr at the end): Old computers often had an "A" floppy drive and a "B" floppy drive, but they shared the same ribbon cable.  Because there was only one cable between the two drives, the computer would use the `Drive Select` pin to activate the proper drive.  However, you'll have noticed that since they share a cable, this requires that the drives actually "know" whether or not *they* are drive A or drive B.  Sometimes there's a small jumper on the drive to allow you to choose, but I've also seen drives without that just assume that **they** are drive A (or drive B).

Tl;dr: Some drives only activate when you ground select-A, some only when you ground select-B.  The Arduino/Moppy doesn't care as long as the drive itself is selected somehow.

NNNenov, 2019-03-06:
Awesome, thanks guys! Can't wait to get started!

closed 2019-03-10 by NNNenov