samphonic/Moppy2 #48: Compatibility with Arduino Pro Micro

NNNenov, 2019-03-10:
Good day, 
I was looking through my stuff and found I only have an Arduino Pro Micro, I was wondering if it would have the correct specs to work with moppy2?
It says its 5v, 16mhz, does that mean it would be using 5v TTL?
And since its so small, I'm wondering if the rest of the specs might be limiting in some way?
I'll try it out since I have it here, but wanted to check just in case.
Thanks!

SammyIAm, 2019-03-11:
It looks like the Pro Mini [has nearly the same specs](https://www.arduino.cc/en/products/compare) as the Uno, so I don't think you'll have any trouble.

closed 2019-03-11 by SammyIAm

NNNenov, 2019-03-12:
thanks, tried it and I can confirm it works with the Pro Micro :-)