samphonic/Moppy2 #49: (Error) Can't upload Moppy.ino to Arduino Uno clone

EdislavNaumov, 2019-03-10:
Hello guys, almost 2 days im trying to make it work without success....I followed everything as described, the floppy is wired to the arduino the right way,it makes some sounds when i powerup the arduino but when i try to Verify and Upload the Moppy.ino in the ArduinoIDE it gives me the following error:

PS. if i select Arduino/Genuino Uno from the Boards menu, it Verifies the code but can't upload it to my "WAVGAT UNO R3" . My "WAVGAT UNO R3" works just fine with other codes(used it for lot of projects without troubles).
[image]

ARDUINOIDE ERROR:

Arduino: 1.8.8 (Windows 10), Board: "WAVGAT UNO R3"

Warning: platform.txt from core 'Arduino AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
sketch\src\MoppyNetworks\MoppySerial.cpp: In member function 'void MoppySerial::readMessages()':

sketch\src\MoppyNetworks\MoppySerial.cpp:75:40: error: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'char*' [-fpermissive]

         Serial.readBytes(messageBuffer + 4, messageBuffer[3]);

                                        ^

In file included from C:\Users\E D\Documents\Arduino\hardware\WAV\avr\cores\lgt8f/HardwareSerial.h:28:0,

                 from C:\Users\E D\Documents\Arduino\hardware\WAV\avr\cores\lgt8f/Arduino.h:205,

                 from c:\users\edeffd~1\appdata\local\temp\arduino_build_317588\sketch\src\moppynetworks\moppyserial.h:10,

                 from sketch\src\MoppyNetworks\MoppySerial.cpp:1:

C:\Users\E D\Documents\Arduino\hardware\WAV\avr\cores\lgt8f/Stream.h:76:10: note:   initializing argument 1 of 'size_t Stream::readBytes(char*, size_t)'

   size_t readBytes( char *buffer, size_t length); // read chars from stream into buffer

          ^

exit status 1
Error compiling for board WAVGAT UNO R3.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

SammyIAm, 2019-03-11:
Looks like some sort of incompatibility between the Serial implementation for the Arduino Uno and the WAVGAT UNO R3.

It'll be tough to try to resolve this without being in front of the board, but it sounds like it's unhappy that I've used `uint8_t` as the type for the `messageBuffer` array.  You could try changing that array to type `char` and see if that helps?  (It's possible it might end up breaking a bunch of other code though since I'm using `uint8_t` rather a lot.

EdislavNaumov, 2019-03-12:
Thank you for your answer Sammy, i found a solution to the problem without changing your array, i installed some drivers i found in the web, in the extension folder for the Arduino IDE.(Documents\Arduino).
I'll attach the ZIP here, if someone has my same problem i'm happy to help.

[Arduino-WavGat-Drivers.zip](https://github.com/SammyIAm/Moppy2/files/2954655/Arduino-WavGat-Drivers.zip) There are instructions on how to install them, in the Readme file inside the ZIP.

SammyIAm, 2019-03-14:
Nice!  Glad you got it working, and thanks for that zip file!

closed 2019-03-14 by SammyIAm