samphonic/Moppy2 #50: 5.25" drive heads colliding with head stops

pootisDan, 2019-03-15:
So I've done up a setup, with both 3.5 and 5.25 drives. The 5.25 drive heads bang against their endstops for awhile instead of immediately reversing, causing a grating rattling BRRRRRT in the middle of my song(s). This doesn't happen for the 3.5" drives, and doesn't happen at all on one or two songs.  
Am I doing something wrong? The DIR pin is definitely connected, since the drives DO reverse direction, but they're always doing it quite a bit after the head hits the end.

SammyIAm, 2019-03-15:
A lot of the 5.25" floppy disks had less tracks than than the 3.5" disks.
In the Arduino code, switch the maximum position for that drive from 158 (I
think) to 78 instead.

(On my phone so I can't see the code, but let me know if that's not clear
and I'll look it up later)

Sammy1Am, 2020-08-12:
Closing for lack of activity.

closed 2020-08-12 by Sammy1Am