samphonic/Moppy2 #52: Using Moppy Control GUI

NNNenov, 2019-03-17:
Hello, I've managed to sort of get some music going with the Moppy control GUI, but I don't really understand whats its doing, sometimes it doesn't really seem to work.

1. i) How does MIDI In work? I have loopMIDI set up and have set the MIDI In option to my midi channel, in my DAW I am outputting live midi to that same channel, but I'm not getting anything out of the FDDs. 
   ii) I noticed the MIDI In option is greyed out unless you load a file first, why is that?

2. i) How can I specify which midi channel is played on which device? When I set the "One Device per Channel" option, it generally tends to not work at all, but I do get results if I set it to Device 1. 
 ii) in a multi track midi file, I'm not really sure whats going on, because there are channels and instruments, are they different things? How would I go about specifically setting which instrument is played on which FDD?
iii) What is a sub Address?

3. I think I just generally don't understand the Mapper at all, is there any documentation on it anywhere? Ideally I would like to use MIDI in feature and understand how to target specidic FFD with specific channel from my daw.

Regardless of all this I've still been having fun with some random files, thank you for sharing this great tool set!

Best,
Nick

NNNenov, 2019-03-17:
Apologies, after thinking about it a bit more I think I understand how the mapper works;
- you select the channel (and other options like note range) in the Condition section
- Device Address is where you set which device to route the data to, e.g. using multiple arduinos
- Sub Address is where you define which floppy drive/device to send the data to
- note is where you can do note operations, 
it would be nice to know what kind of code/functions are available to use in these fields, I noticed reading it can be set to random, but I don't know what code to use to do that.

That nullifies most of my earlier questions, but I still haven't figured out how the MIDI In feature works

NNNenov, 2019-03-17:
Ok, I apologise once again, I got MIDI In to work, I was just not configuring my DAW correctly.
So everything is working perfectly, thank you!

closed 2019-03-17 by NNNenov

SammyIAm, 2019-03-18:
Hey, no need to apologize, glad you got it working!

There's [very light documentation on the wiki](https://github.com/SammyIAm/Moppy2/wiki/Getting-Started), but it could definitely be more in-depth.

> it would be nice to know what kind of code/functions are available to use in these fields

It's Javascript, so any of the built-in Javascript functions are available.  The only predefined bits available are `c` for channel, `n` for note (both from the MIDI file), and the `nextRoundRobinSubAddress(n)` function that's defined in [MIDIScriptMapper.java](https://github.com/SammyIAm/Moppy2/blob/master/Java/MoppyLib/src/main/java/com/moppy/core/events/mapper/MIDIScriptMapper.java).

You can add your own functions by defining them in `MIDIScriptMapper.java`.  (You can also define them directly in the mapper box, but the code in that box is evaluated for every single MIDI event, so it might be a tad bit inefficient to continually redefine functions)