samphonic/Moppy2 #53: A3967 support

tobiaslp12, 2019-03-18:
Hi,
I want to use an unipolar stepper motor in a scanner. I have the stepper connected to an EasyDriver (4.4) but i have no idea on how to make the instrument code. Has anyone done this or are there just some small adjustments to the L298N code?

Thanks in advance

Tobias

(btw. I like the new version of moppy; Can you add to the wiki to open the .bat file with admin perm. bc i had to do it for it to work)

tobiaslp12, 2019-04-09:
Hi, 
so i modified the floppy code a bit and my scanner is now playing notes(not sure if correct though; but YAY).
The only thing that is not working is the reset function. It technically works but i deactivated it, because it moves the scanner head backwards, even if it is already at the end, which means the rubber band (which moves the scanner) skips. 
How do I modify the reset function to only move the amount, it moved forward, backwards?

owenr930, 2019-04-10:
I have success with eh floppy drive so far but I don't know how to do the
same thing with scanners. Could you let me know how you are doing it?

tobiaslp12, 2019-04-10:
Hi I connected the wires of the stepper motor to the Easy Driver by Brian Schmalzhaus(1 coil to the a pins and the other coil to the b pins). To power the stepper motor and the driver I currently use a lab power supply(I can change the volt etc). Be sure what power input your stepper needs. I am planning to connect a pc psu rather than the lab psu to the GND and M+ pins to power the stepper and driver. The driver can be connected as a floppy drive to the arduino and the GND pin to GND on the arduino. In my code I used 2 more pins for the speed control. Here is a simple pin table:

| Arduino pin | Driver pin       |
|-------------|------------------|
| 2           | Step             |
| 3           | Dir              |
| 4           | MS1              |
| 5           | MS2              |
| GND         | GND next to Step |

I need to make the code a little clearer so i'll post it when it is (probably tomorrow). It still has some problems like:
> How do I modify the reset function to only move the amount, it moved forward, backwards?

I hope you could understand how i got it to work and I could help you.

tobiasfrck, 2020-07-25:
Well, I finally did it and created a pull request (my first, yay) #105. The EasyDriver instrument needs two switches for direction changing and reset functions. The reset function is disabled by default because stepper motors can take too long to reset fully which is a bit annoying. Another reason is that by using these switches the position of the stepper motor is not important as it will always change direction if a switch is triggered. If you want to use the reset function, just uncomment the parts I commented with: "Uncomment this if you want to be able to reset the drivers!". Also make sure the variable max_position is big enough for all stepper motors connected to the arduino to fully reset.

Sammy1Am, 2020-08-16:
This has been merged into master now, and should be ready to test.  I haven't uploaded a binary of the Java application yet, but the new Arduino code should be backward compatible in that regard so you should be able to just grab the source for the Arduino code, open it in the IDE, and upload it.

If you get a chance to check it out, let me know how it goes so I can fix any issues.

tobiasfrck, 2021-07-16:
Hi,
so this has been a while since i've tested my moppy setup and I apologize for not answering you.
I've now tested the update and was confused for a second that the arduino sketch only contains one tab and I had to take a close look at the settings at first. Now it's working again and I haven't had any problems with my setup. 
It's just that my setup is kind of out of tune and it's hard to find a song that the scanner can actually play.
There is one song that i found works pretty well but it's a song i'd hate if i heard it on the radio :D
I'd love to post a sample of it here but i don't know if it's according to copyright :(

So thanks for helping me out with this issue and my first ever commit to a github project :)

I think this issue can be closed now.

Sammy1Am, 2021-07-17:
Glad you got it working!

For what it's worth (I am not a lawyer), you should be okay posting a sample.  YouTube's been pretty okay with copyrighted songs existing, I just can't monetize them.  Would love to hear/see a sample though!

closed 2021-07-17 by Sammy1Am

tobiasfrck, 2021-08-19:
So it took me a while to make the third demo but here are three demos of nine floppy drives with a scanner (driven with the easydriver):
https://www.youtube.com/watch?v=ajrwglWmFBw
https://www.youtube.com/watch?v=BoVm3tkzqJI
https://www.youtube.com/watch?v=IlVLHYT7eSc
I got copyright claims on two of those videos but it's fine since they are just claims.
Oh and sorry for the quality. I couldn't do much about the fan noise and they were just supposed to be some demos.

Sammy1Am, 2021-08-23:
Nice!  It sounds great!

And yeah, I don't know if I'm more impressed by the YouTube algorithm or the floppy drives that one of them is so good it recognizes the songs.  Most of my videos got copyright claimed (even the public domain ones 😠 ), so that's a shame.