samphonic/Moppy2 #54: Multiple Floppy Problem

shaikhs234, 2019-03-21:
Greetings,
I was using Moppy 2.0 for one of my projects and I have 8 floppy disks at the moment, all working perfectly fine because the light is always turned on when I connect the drive select pins and the power supply. 

But the issue is, when I try to play the Kirbytheme song for example, only 3 or 4 drives play it at times and it's usually not audible. I was wondering how to get it working?

And also if I try to download another MIDI file, most of the times it does not play it properly. Can anyone guide me to a site that has MIDI files suitable for my configuration?

SammyIAm, 2019-03-21:
Hello there!

The Kirby theme midi only has 3 or 4 MIDI tracks, so only the first few drives will play.  You can duplicate the events from those tracks by creating a second event mapper in the Moppy GUI and setting the SubChannel script to `c+5`; this will shift the channel numbers up (e.g. from 1 to 5) so that the notes will be duplicated on the higher numbered drives.

Unfortunately you're unlikely to find MIDI files designed to work with Moppy on most websites.  You might have some luck asking other musical floppy creators if you can have their MIDI files, though.  It might be cool if a website for floppy MIDI files *did* exist though... maybe you could make one!  😉 

For the best performance from arbitrary MIDI files found in the wild, so to speak, your best bet is to set the SubAddress script to "Round Robin", and the Note script to "Force into Range".  This attempt to play all of the notes present in the MIDI file on the floppy drives, which potentially will sound like a mess, but it'll be close.  You may also want to use `c!=10` as the Condition script to ignore MIDI drum events (since they're not pitched-notes the same as the rest)

shaikhs234, 2019-03-21:
Thank you so much for the quick reply! I understand what you mean and it does make sense :) I will try doing what you said!

SammyIAm, 2019-03-21:
Let me know how it goes!

shaikhs234, 2019-03-24:
It works perfectly fine! I found some Moppy MIDI files on github and they sounded amazing. Thank you so much for saving the day!! Also, if you could find the moppy midi file for megalovania Undertale I'd appreciate it :D

SammyIAm, 2019-03-24:
Glad it worked!  I don't have that one, but if you've seen it somewhere, you could always ask whoever recorded it if they would be willing to share.

closed 2019-03-24 by SammyIAm