samphonic/Moppy2 #55: Music Player Moppy

PablooAjofrin, 2019-03-21:
Good afternoon, I'm Spanish and I'm studying Robotics at my highschool so I'd like to create a project with this tool, I'd like to create a type of music player using the floppy drives, also with a LCD screen and some buttons in order to change the different songs, I don´t know if that could be possible because of Arduino's memory (probably low to store all the MIDI files), and I don't even know if the MIDI is played directly from the PC or stored at some moment in the Arduino.
I'm improving my English, sorry!

SammyIAm, 2019-03-21:
Don't be sorry for improving your English 😉 , I've forgotten nearly all of my high school Spanish by now!

I have seen other projects on the Internet (though I don't have any links unfortunately) that did use Moppy with an LCD screen to show the currently playing song, so at least that much is possible.

The Java software on the PC converts the MIDI file into MoppyMessages (smaller representations of the MIDI notes) because there is a lot more computing power on the PC.  They do have Arduino shields that allow you to use an SD Card, but there is currently no support in Moppy for reading the MIDI files directly on the Arduino.

A possible option for you though: the communications between the PC and Arduino are bidirectional, so you could potentially have the buttons on the Arduino send a message to the PC to change the song.  It would still require that the Arduino be connected to a PC, but the PC could be something small like a Raspberry Pi.

If you decide to go ahead with the project and run into specific issues, definitely post here and we'll see what we can do to help.

PablooAjofrin, 2019-03-21:
I'll post the future changes I'm doing with the project, thank you very
much for the help, maybe I try to use just one song and that's it!

DJthefirst, 2019-03-28:
this reminds me much my onetesla interrupter http://onetesla.com/products/kits/interrupters-all/sd-card-interrupter-kit.html. it uses a sd card to hold the files and you are able to select tones  and  songs with two buttons. one tesla actuall uses midi files also and sends the data through fiber optic.

DJthefirst, 2019-03-28:
The circit designes are downloadable on their website im not sure about the code though because they use just the arduino atmel chip.

Sammy1Am, 2020-08-12:
Closing for lack of activity.

closed 2020-08-12 by Sammy1Am