samphonic/Moppy2 #56: Power Supply Dying

TheFuzzLlama, 2019-03-21:
I got it to work, but I was only using 4 drives with 2 separate power supplies, now I’m trying to use 6 drives with one power supply. Now the power supply randomly shuts off, I’ve used a power supply tester to verify that it does work, I’ve noticed uploading code and loading the drives in a MoppyControlGUI causes them to die. The power supply runs at 250W, which should be enough to power these 3.5” drives. I think it’s a communication error somewhere.

SammyIAm, 2019-03-21:
Connecting to the Arduino in the MoppyControlGUI causes all of the drives to reset, so that would cause a bit of a power usage spike.

What kind of power supply are you using?  If it's an AT/ATX supply, the 250w is for the whole of the supply; the 5v rails (which is what the floppy drives are using) are usually rated for a much smaller amount since the majority of a normal desktop's power consumption is 12v.  It sounds like 6 drives might be too many.

If you're willing to do a little wire splicing, you can get (or find maybe) fairly cheap 5v power supplies you could use instead of a computer power supply.

TheFuzzLlama, 2019-03-21:
It’s ATX. I’ll look at them.

TheFuzzLlama, 2019-03-21:
Edit: They’re all on seperate rails, I spliced them from other ones, like
the 20 pin mobo cable and the cpu power cable.

SammyIAm, 2019-03-21:
They may be on separate wires, but internally it's likely that all of those 5v wires are coming from a single 5v power source (which has a maximum power output much lower than 250W.  Probably more like 130W.).  You should be able to see how many Watts/Amps are available on the 5v rail by looking at the sticker on the PSU.

(Tangential, but a multi-railed PSU has a [sticker like this](http://modularpower.computer-parts-store.com/images/Platimax_750W_4_large.jpg) that shows multiple 12v outputs)

Sammy1Am, 2020-08-12:
Closing for lack of activity.

closed 2020-08-12 by Sammy1Am