samphonic/Moppy2 #57: Direct Midi Input

MagnusM1709, 2019-03-31:
Is there a possibility of using direct Midi Input at the Arduino?

So I can attach a Midi keyboard directly without an PC.

MagnusM1709, 2019-04-02:
Thank you for your information.
I think I will update the moppySerial.h and moppySerial.cpp, but I am a little bit confused how the systemHandler and deviceHandler work.

Can you explain me how they work and which parameters I have to provide (e.g tone (c1) or tone of(c1))

Thank you for your help

Magnus Müller

SammyIAm, 2019-04-02:
You can see the implementations for handling `systemMessage` and `deviceMessage` in [FloppyDrives.cpp](https://github.com/SammyIAm/Moppy2/blob/master/Arduino/Moppy/src/MoppyInstruments/FloppyDrives.cpp).

In both cases, the second-to-last parameter is a command that specifies what sort of information will be contained in the last parameter.  For example, if the `command` parameter is `NETBYTE_DEV_NOTEON` then the `payload` is the note to turn on.  The contents of these commands and payload is described on the [wiki here](https://github.com/SammyIAm/Moppy2/wiki/Design-Docs).

The deviceMessage handler differs from the system handler in that it also has a `subAddress` parameter that specifies which of the drives controlled by that single Arduino should have... whatever the command is done to them.

There's generally a fairly 1 to 1 mapping of MIDI message event types to MoppyMessage commands, so you'll mostly need to convert the appropriate MIDI events into the values expected by MoppyMessages (for which the Java code is probably a good example, since it does just that)

MagnusM1709, 2019-04-02:
Thank you for your answer, but I have another Question:
When is the **void MoppySerial::readMessages() ** function called? Periodically by a timer? or if there is serial data available?

SammyIAm, 2019-04-02:
If your IDE won't find references for you (and some definitely don't do a great job), GitHub has a handy [search in repository](https://github.com/SammyIAm/Moppy2/search?q=readMessages&unscoped_q=readMessages) feature that makes this fairly easy to find.

Though you can also see that the readMessages implementation returns immediately if there isn't any serial data available to avoid locking up the Arduino.

MuffyTarkin, 2019-04-03:
I'm getting back into working with Moppy again, and I can tell you that on MoppyClassic it was possible to do this by way of a loopback MIDI program, and something like FL Studio. I haven't done any work with Moppy 2 yet (just now looking into it), but MoppyClassic featured a setting where you could tell the Java GUI what MIDI device/port to listen on for MIDI input. By default, this would simply look for a file selection (load in a MIDI file from the main GUI, and it would play it back).  

A MIDI loopback program on the other hand would create another MIDI device on your computer, a virtual one, if you will. The one I used would show up in Moppy as a listenable device. From there, I could open up FL Studio and direct its MIDI output to that virtual MIDI port created by the loopback. Voila, I could directly play MIDI tracks in FL Studio and instantly hear them being played back on the drives.  

I personally don't have a MIDI keyboard, but I know for a fact that FL Studio allows you to hook up a MIDI instrument and use it within the program. So you could, in theory, use a similar loopback procedure to get Moppy to accept MIDI data from FL Studio, and simply add all the necessary tracks and connect all of the individual channels to one layer, and play notes using the layer channel to get all of the drives to play those notes.

MagnusM1709, 2019-04-04:
Hello,

i have a problem with the  **deviceHandler**. I have the problem, that the function is _never_ executed.
The lines of code with _hardcoded_ values should work (I think), but they don't:
`
uint8_t payload[] = {70, 127};
deviceHandler(0, 0x09, &payload[0]);
`

i have build a Serial statement before and after the Method, they get both print on the Console and a Serial print statement at the beginning of **deviceMessage** in class **FloppyDrives.cpp**, but this statement is never printed on the Serial Monitor.

Where do I do a failure?
Please Help

Thank you,
Magnus Müller

SammyIAm, 2019-04-04:
Hi @MagnusM1709 , would you mind posting a bit more of you code for context?  It's hard to tell what might be going on from just a single line.  Ideally if you've forked the Github repo you could just push your changes to your own repo and post a link to it.  Failing that, pastebin is probably a better choice over putting the code directly in here (but if all else fails that'll work too).

@MuffyTarkin Moppy 2 does support similar MIDI In (and Out!) functionality as the classic version, so hopefully setting it up is fairly straight forward.  It appears that Magnus's question is about sending MIDI directly to the Arduino and skipping the PC entirely-- so related, but a bit more involved on the Arduino code side.

MagnusM1709, 2019-04-05:
### MoppySerial.cpp
```
#include "../MoppyNetworks/MoppySerial.h"

/*
 * Serial communications implementation for Arduino.  Handler
 * functions are called to consume system and device messages received from
 * the network.
 */
MoppySerial::MoppySerial(handleSystemMessage sys, handleDeviceMessage dev) {
    messageBuffer[0] = START_BYTE;
    systemHandler = sys;
    deviceHandler = dev;
}

void MoppySerial::begin(long baud) {
    Serial.begin(baud); // need to be 31250 for midi
}

/* MoppyMessages contain the following bytes:
 *  0    - START_BYTE (always 0x4d)
 *  1    - Device address (0x00 for system-wide messages)
 *  2    - Sub address (Ignored for system-wide messages)
 *  3    - Size of message body (number of bytes following this one)
 *  4    - Command byte
 *  5... - Optional payload
 */

void MoppySerial::readMessages() {
    Serial.println("Start device playing");
    uint8_t payload[] = {70, 127};
    deviceHandler(0, 0x09, &payload[0]);
    Serial.println("Device Played?");
    
}

void MoppySerial::sendPong() {
    Serial.write(pongBytes, sizeof(pongBytes));
}
```

### FloppyDrives.cpp 
##Only deviceMessage, because I changed something here
# Changes: Adding a Serieal.print statement

```
void FloppyDrives::deviceMessage(uint8_t subAddress, uint8_t command, uint8_t payload[]) {
    Serial.println("Command: " + command + "\t NetByte: " + NETBYTE_DEV_NOTEON);
    switch(command) {
        case NETBYTE_DEV_RESET: // Reset
            if (subAddress == 0x00) {
                resetAll();
            } else {
                reset(subAddress);
            }
            break;
        case NETBYTE_DEV_NOTEON: // Note On
            // Set the current period to the new value to play it immediately
            // Also set the originalPeriod in-case we pitch-bend
            Serial.println("Right Payload");
          
            if (payload[0] <= MAX_FLOPPY_NOTE) {
                Serial.println("schould play");
                currentPeriod[subAddress] = originalPeriod[subAddress] = noteDoubleTicks[payload[0]];
            }
            break;
        case NETBYTE_DEV_NOTEOFF: // Note Off
            currentPeriod[subAddress] = originalPeriod[subAddress] = 0;
            break;
        case NETBYTE_DEV_BENDPITCH: //Pitch bend
            // A value from -8192 to 8191 representing the pitch deflection
            int16_t bendDeflection = payload[0] << 8 | payload[1];

            // A whole octave of bend would double the frequency (halve the the period) of notes
            // Calculate bend based on BEND_OCTAVES from MoppyInstrument.h and percentage of deflection
            //currentPeriod[subAddress] = originalPeriod[subAddress] / 1.4;
            currentPeriod[subAddress] = originalPeriod[subAddress] / pow(2.0, BEND_OCTAVES*(bendDeflection/(float)8192));
            break;
    }
}
```

### Serial Output:

> 16:57:06.224 -> Start device playing
> 16:57:06.224 -> Device Played

### Issue:
Apparently the function deviceMessage is never executed

What am I doing wrong??

MagnusM1709, 2019-04-05:
Hi,
I think I figured out, why the code wasn't working:
\t I worked in the wrong workspace, so my changes did nothing
So forget the last two messages

SammyIAm, 2019-04-05:
Cool, glad you solved that mystery at least!

MagnusM1709, 2019-04-06:
Hello,

I have now written my first version of a moppy repository with direct midi input, but I don't want to make a pull request because there are many bugs and not very good code and efficiency.

I've sent you a collaboration request for my private repository so you can do a check (if you like).

The software should have polyphonic support (useing more than 1 drive), but I can't test it because I now have only one drive.

When I have tested the code and made some improvements, I will start a pull request.

If you have additional ideas, I will probably implement them.

cbmainz, 2020-02-18:
hi @MagnusM1709, so could I use your fork and have direct Midi-In support?

Cheers
Christopher

cbmainz, 2020-02-18:
Okay, I found the answer myself. Sorry. 

I will test this in some days after receiving some floppy disk drivers. It has been some time were I builded the first Mobby instillation with some kids.

Sammy1Am, 2020-08-16:
I'll be honest, I don't remember exactly what the outcome here was, but there appears to currently be MIDI support in the master branch.  Closing for now (since I must have pull this somehow in the past), but feel free to reopen if I missed something (or open a new ticket if there's something new and broken)

closed 2020-08-16 by Sammy1Am