samphonic/Moppy2 #59: Consecutive channels in two arduinos (1-8, 9)

ComputerSongs, 2019-04-13:
Hi. I have been working with a configuration of 8 units with arduino uno, and now I want to add a set of scanner, l298n and arduino nano, which works perfect but I do not happen to make it work in parallel with the floppy drives.
What I'm looking for is to separate the l298n from the diskettes, and put it on channel 9. In this I did not do very well. I assigned the address 0x02, configure minsubaddress and maxsubaddress in 9, in Moppy I added a mapper with device 2, it did not work in the end. While the floppies sounded, the scanner did not.
Thanks for the help.

SammyIAm, 2019-04-16:
Can you post a screen shot of the MoppyGUI mappers you're using?  Adding a second mapper with device address set to 2 should have worked.

ComputerSongs, 2019-04-18:
[image]

Sorry for the delay I hope it serves you

SammyIAm, 2019-04-24:
No problem, I too am sorry for my delay in responding.

It looks like you do have the device address correct.  The Sub Address on your second mapper still shows c+1, which means that events sent to device 2 will still have a sub address assigned based on the channel of the note being played in the MIDI file.  Because you've set the min and max sub addresses both to 9, the device code will filter out any messages is sees on channels other than channel 9.

- Does your MIDI file have events on MIDI channel 9?
- What happens if you change the Sub Address script for your second mapper to just `9`?
- On the Arduino Nano, which pins are you using for the L298N?  The device code maps sub-address values to pin numbers (e.g. 1->2&3, 2->4&5, etc.).  If you're using pins 2 and 3, you'll probably still want to use sub address 1 for the device (change the min/max back to 1, or even 1-8 would be fine).

ComputerSongs, 2019-04-24:
Hello again. I have already tried changing the sub address to 9 in the second mapper in the program, with the minimum/maximum subaddress set to 9 on the Arduino. Try changing this setting to 1, and do not get results. The midi has the melody set to channel 9 with a range of notes acceptable to Moppy. Now I am thinking that the Arduino could be using a non-configured pin. Let me explain, the code L298N.cpp declares a configuration for four engines, and the program may be sending messages to be reproduced in engine 9, which according to this file is not configured.
Maybe you can try with an Arduino Mega and try to configure the missing pins until you reach the ninth, and in this connect the L298N, as long as you do not have to make many changes.
It is my opinion and maybe a possibility.

SammyIAm, 2019-04-25:
Ah, okay, I looked over the L298N.cpp code a bit, and it's definitely expecting only to have messages for sub-addresses 1,2,3, or 4.  So here's what you'll want to do:

- Change your `MIN_SUB_ADDRESS` in the Arduino code to `1` and `MAX_SUB_ADDRESS` to `4` (I know you only have 2 things connected, but it should't cause any issues.
- Set your mapper scripts for the second mapper to the following (in order from left to right)
  - `c==8` (Only map notes on channel 9)
  - `2` (Device address 2)
  - `c-7` (This is a little confusing, but channel 9 has a `c` value of `8` (because they're 0 indexed), and you want to sent events coming from channel 9 to sub-address `1` on the Arduino.  (8-7=1)  This will also make channel 10 map to sub-address 9 for when you add another device to that Arduino)
  - `n`

I *think* that should do it for you.  I can see the confusion arising from the way the subaddresses are sort of implicitly matching channels, so I might try to clarify that in the documentation next time I'm making changes.

ComputerSongs, 2019-04-25:
Thank you. I configured as you said and it worked, besides, it was not even necessary to place c == 8 in the first place, it was enough with cologar c-7 and it worked, The truth is that I have no idea of javascrip, so I did not know well how to configure these lockers.
  Thanks for your help. By the way, if you want to use up to 16 devices, what would be the configuration to perform?

SammyIAm, 2019-04-25:
You should be careful setting the first column to `true` just blindly.  In this case, it's still processing notes from e.g. channel 3, and passing them along to sub-address `-4` (which may end up mapping to `250` depending on how overflows work).  You may not be seeing any weird behavior at the moment (or may never), but there's always a chance later on you'll be trying to do something and mysteriously start getting negative sub-addresses.

There's no upper limit for the number of devices you can use, but there *is* an upper limit on MIDI channels of 16.  Just make sure that you're mapping each channel to a different address/subaddress combo and you shouldn't have any trouble.

ComputerSongs, 2019-04-25:
Ok. Thanks, I close

closed 2019-04-25 by ComputerSongs