samphonic/Moppy2 #61: Midi-In

JulienLQ, 2019-04-24:
Hello
Could you explain to me the midi-in option

SammyIAm, 2019-04-25:
Sure!

The MIDI protocol is designed for allowing digital musical instruments to communicate in real-time, so that they can all play music together.

The "traditional" Moppy setup involves the MoppyGUI Java application loading a MIDI file, and playing it back for the floppy drives.  MIDI files are interpreted by a "sequencer" that reads the file and generates MIDI event messages at precise times (e.g. note messages in the rhythm of the song).

The MIDI-in option allows you to receive these MIDI events from a MIDI port on your computer instead of from a sequencer interpreting a file.  Unless you have a MIDI hardware device, you'll probably need to get a virtual MIDI cable program, and then you can hook up a program that supports MIDI-out (like [VMPK](http://vmpk.sourceforge.net/)) to pass MIDI events through the MoppyGUI on to the drives.

ComputerSongs, 2019-05-02:
Hello again.
I am dealing with the aforementioned program, but this does not appear as a MIDI input device in Moppy, if that is how it should work, that is, I can not find a way to link both programs. Thank you for your answer.

ComputerSongs, 2019-05-02:
Please, ignore what I said. I just had to download a program that established virtual MIDI cables, configure both VMPK and Moppy on the same channel and that's it.

closed 2020-08-12 by Sammy1Am