samphonic/Moppy2 #62: I²C support

AmEv7Fam, 2019-04-28:
I'm looking at options for allowing multiple hard-wired connections without resorting to using multiple USB-serial adapters (including the ones on Arduino boards). While it's _technically_ possible to hook up all of the serial In ports together and have a common "input", things can get messy when trying to push updates to an individual board. Really, UART is designed for one host and one device connection. SPI and I²C both allow for (basically) an infinite number of connections, but there's one drawback to SPI: you have to have a physical line for each and every "host" device, making expansion that much more difficult. I²C is a happy medium, allowing for enough bandwidth to play even the craziest of songs on the craziest of devices, while being flexible enough to allow upgrades to the "network".

SammyIAm, 2019-05-01:
I2C is a pretty good option, and converts well from Moppy Messages since they follow the same DeviceAddress-DataBytes structure.  I seem to remember someone else working on this a bit, but I can't find any evidence of their code.

Off the top of my head you'd have two options for getting this working with Moppy (though the first one is probably more reasonable):

1. Write an Arduino (or microcontroller of your choice) sketch to receive the Moppy Messages over serial (as normal) and convert them to I2C messages to be sent out from some pins on that board.  The only thing you'll have to look out for is that you'll want the code to receive messages addressed to *any* device, not just one, and currently there's some code in there to ignore messages sent to device addresses other than the device's own.
2. Figure out how to get Java to talk to a USB->I2C device and use that.  This would allow you to talk directly from the PC to the I2C bus, but I'm not sure how the Java library support for those devices would be.

Either way, the receiving devices would have basically the exact same code, except instead of the MoppySerial implementation, you'd implement a MoppyI2C network to interpret those messages instead.

DJthefirst, 2020-06-09:
I was talking to Lothean about it back in 2018 with the issue [24](https://github.com/Sammy1Am/Moppy2/issues/24#issuecomment-416075332) though I closed the project as I got to busy with school to work on it and didn't have all the pieces to the puzzle.

Sammy1Am, 2020-08-12:
Closing for inactivity for now.

closed 2020-08-12 by Sammy1Am