samphonic/Moppy2 #63: Note Velocity

wnycomputerguy, 2019-04-30:
I am entirely self-taught with the exception of some basic C++ classes in college 25 years ago, so bear with me - I am using/modifying MOPPY to generate sound on piezo-based sound devices and would like to add note velocity into the mix. 

I've successfully modified the code so it will work on the Teensy 3.6 (had to change Timer stuff) so it allows a tick resolution of 1, and I've modified the note data to be precise to 4 decimal places  - so sound reproduction on a piezo or even a speaker is pretty much perfect. The Teensy was required because even the Mega seems to give up at tick resolutions below 10.

The modifications are probably hacky and wrong, but they work. I have no clue how forking works, I basically just edit everything in Notepad++ and experiment via trail and error.

Again...amateur. 

Anyway, I've stared at the code long enough to arrive at the conclusion that the velocity data (data2?) is passed to the Arduino and I'd need to modify the instrument code in order to get it to process the velocity data in the loop. 

Am I on the right track?

Thanks....

wnycomputerguy, 2019-04-30:
If anyone cares, in order to get Moppy2 working on the Teensy I had to change the timer library to Timer3. I don't remember exactly why, but it had something to do with the PWM pins I think....it was a guess that worked :) I also removed most of the direction code because it's not relevant to my application. although it worked without removing it. I did it mainly to eliminate unnecessary code so I had less to stare at when trying to figure out how to move forward....

I'm building a display made entirely out of old fire alarm equipment and am trying to stick with the theme of using only the original circuits in the alarm devices rather than gutting them and sticking speakers or piezos in - because that's no fun...the sound is very buzzy and loud as one can imagine, so the ability to modify volume via the midi velocity data is essential in order to not make one's ears bleed. 

I've been able to successfully equalize the volume on all devices using trimpots (they are not all the same model) and have 9 running simultaneously and reliably. I'm also attempting to add percussion via relays that trip a bell and a chime, but that's a project in and of itself...one thing at a time.

SammyIAm, 2019-05-01:
> Am I on the right track?

70% yes, 30% I'm-not-sure 😛 

To start with, yes, velocity is sent as the second byte for noteOn events, so you'll need to modify the instrument code (though ideally you could copy the existing files and create something not called FloppyDrives where you make your modifications) and do something useful with the velocity data (as it's just ignored currently).

The 30% I'm less sure on is increasing the resolution so significantly (though if it's working for you, there's no need to undo it).  How off a note will end up depends a bit on what sort of frequencies you're looking at playing, and floppy drives were low enough it didn't matter much, so it's more important if you're going for high notes.  The precision is inherently limited by the fact that the code is using digitalWrite to write out the waveforms a step at a time, and this process itself takes time (on the order of microseconds in the 0.1 to 3ish range on a Mega238 if I had to guess, but I guess a tenth of that with a Teensy 3.6), so trying to control precision any finer than that is going to be incredibly difficult if not impossible.  Additionally, having an interrupt running that often may start to interfere with reading from the serial port (although again, less an issue on the Teensy).  If you're really looking for precision, you'll probably want some sort of discrete hardware timing module that's not connected with the code's main loop.  As I said though, as long as it's working for you, there's no reason to go an undo it.

If you're planning to vary the amplitude of the output signal, you'll just need to convert the velocity into amplitude.  Alternatively you could try to very the duty cycle of the wave, but I suspect timing might start to get kinda tricky with that.

wnycomputerguy, 2019-05-01:
Thank you for the info,  it sounds like I was on the right track... The resolution increase is working perfectly on the Teensy; it may not be the most efficient way to do it but the processor seems to be handling it fine (I don't know how much faster it is, but apparently it's fast enough.) I also modified the note periods to be long instead of int, crunched the numbers in google sheets based on a frequency chart I found somewhere on the internet, updated them to four decimal places, and added in all notes found on an 88 key piano keyboard which should cover my needs.  It looks like this now:

```
const long notePeriods[128] = {
    0.0000,0.0000,0.0000,0.0000,0.0000,0.0000,0.0000,
	0.0000,0.0000,0.0000,0.0000,0.0000,0.0000,0.0000,0.0000,0.0000,0.0000,0.0000,0.0000,
    30581.0398,28868.3603,27247.9564,25706.9409,24271.8447,22904.2602,21626.2976,18181.8182,17161.3129,16198.1618, //C0 - B0
    15289.0237,14430.8890,13620.9719,12856.4780,12134.8915,11453.8091,10810.9744,10204.1858,9631.4989,9090.9091,8580.6712,8099.0809, //C1 - B1
    7644.5119,7215.4549,6810.4860,6428.2473,6067.4531,5726.9111,5405.4872,5102.0981,4815.7494,4545.4545,4290.3356, 4049.5339, //C2 - B2
    3822.2501,3607.7379,3405.2523,3214.1319,3033.7229,2863.4588,2702.7465,2551.0464,2407.8747,2272.7273,2145.1678,2024.7669, //C3 - B3
	1911.1250,1803.8624,1702.6203,1607.0608,1516.8614,1431.7294,1351.3733,1275.5265,1203.9345,1136.3636,1072.5839,1012.3855, //C4 - B4
	955.5643,901.9328,851.3102,803.5304,758.4319,715.8647,675.6857,637.7624,601.9723,568.1818,536.2919,506.1922, //C5 - B5
	477.7831,450.9664,425.6551,401.7645,379.2159,357.9329,337.8424,318.8816,300.9836,284.0909,268.1454,253.0966, //C6 - B6
    238.8915,225.4832,212.8275,200.8823,189.6080,178.9658,168.9212,159.4408,150.4918,142.0455,134.0731,126.5480, //C7 - B7
	119.4455,112.7416,106.4140,100.4413,94.8040,89.4831,84.4607,79.7203,75.2459,71.0227,67.0365,63.2741};//C8-G8
```

(Note that the compiler does complain about some conversion issues, but it works so I haven't yet bothered to get into that one yet...)

I have already created a separate instrument apart from Floppy and have that working, so I'm guessing I have to play with some math and horrendously complex number crunching in order to make it work.

Am I correct in assuming the variable that I'd need to modify in the .h file is:

`static void deviceMessage(uint8_t subAddress, uint8_t command, uint8_t payload[]);`

so it would need to be:

`static void deviceMessage(uint8_t subAddress, uint8_t command, uint8_t payload[] ,uint8_t payload2[] );`

and then the math would be handled in the .cpp file:

```
    case NETBYTE_DEV_NOTEON: // Note On
      // Set the current period to the new value to play it immediately
    	// Also set the originalPeriod in-case we pitch-bend
      if (payload[0] <= MAX_FLOPPY_NOTE) {
        currentPeriod[subAddress] = originalPeriod[subAddress] = noteDoubleTicks[payload[0]];
      }
      break;
```

Am I still on the right track?

Thanks again for the  original prompt reply, much appreciated!

SammyIAm, 2019-05-02:
In 
`static void deviceMessage(uint8_t subAddress, uint8_t command, uint8_t payload[])`
the `payload` argument is already an array.  So you won't be adding a second payload array, you'll just want to access the second byte in the payload (see the `NETBYTE_DEV_BENDPITCH` case for an example of referencing `payload[1]`).

You're probably right on where the math would be handled though.  One potential option is to create an array similar to `currentPeriod` called `currentVelocity` and set the values there similarly.  Then in `FloppyDrives::tick()` you can read the velocity and set the appropriate output information on whatever pins you're using.

closed 2020-08-12 by Sammy1Am