samphonic/Moppy2 #64: dot matrix printer instrument - please help

Chong-McBong, 2019-05-01:
hi, i have built a dot matrix printer instrument out of an Epson LQ-300+, but am completely stuck with the code, can someone help me put something together?

i have an arduino mega, 24 outputs connected to 24 dot matrix pins (via the printer's onboard hardware) - the plan is to send it Round Robin data, so it picks the next free pin, and uses all 24. 
it works, i can click the pins manually with 5V, just need some help with the code.

i am planning to just send it data as if it was 24 floppy drives, without direction pins. so the dot-matrix pins fire at audio frequencies. i might also need an option to change the length of the pulses to stop it overheating the print head.

it might possibly end up being run from an uno, with the pins doubled up to make 12 driveable pins. depends on volume

i will share circuit diagrams and stuff on the wiki if it works :)

there are also a paper up/down motor, and head left/right motor that i plan to control so it prints mess as it plays - they aren't wired up yet though

Chong-McBong, 2019-05-01:
small update, i managed to edit a code to make all the pins fire in sequence, so i know they all work. my coding skills are not good enough to figure it out as an instrument though

SammyIAm, 2019-05-02:
Glad you figured out how to open the ticket again 😉 

Without actually doing things myself, I can't give you _precise_ instructions, but I can certainly point you down the right path.  The good news is, the basics should be pretty straight forward (adjusting the pulse length will be very tricky though).

To start, if you connect up the step pins to a few of the dot matrix pins instead of a floppy drive, you should be able to test with at least the first 8 pins to make sure things are working normally without making any code changes.  This'll give you a good baseline to start from as you start playing with the code.  *NOTE:* The FloppyDrive code will leave a pin in the ON state after a note has stopped playing (about 50% of the time) so if you're worried about overheating, just make sure it's playing notes all the time 😜 (or unplug it as soon as the test works).

1. Copy `FloppyDrives.h` and `FloppyDrives.cpp` to new files called ... `DotMatrix.h` and `DotMatrix.cpp` (or whatever you want to call them).  In `MoppyCore.cpp` change any references to `FloppyDrives` to `DotMatrix`.
2. Your first code changes will be in the `FloppyDrives::tick()` and `FloppyDrives::togglePin` functions.  As it is, there are 9 blocks in the `tick()` function that each check if that subaddress/drive/pin is supposed to be playing a note, and toggles on or off the pin if it is.  In your case, `togglePin` only needs 2 arguments (since there's no step pin), so you can remove that second argument.  `tick()` then can call `togglePin` pretty much as it's doing, except instead of `togglePin(1,2,3)` and `togglePin(2,4,5)`, you'll have `togglePin(1,2)` and `togglePin(2,3)` (assuming you have your matrix pins connected to pins 2 and 3 on the Arduino.

At this point, rewire your matrix pins to 2,3,4,5,&c. on the Arduino and try again to make sure that worked at least up to pin 9.  If it doesn't work, fix that first, but then you're good to expand your pin count and make some other changes.

3. As mentioned above, if you want to make sure the pin is off after a note goes off, you'll need to modify the `NETBYTE_DEV_NOTEOFF` case to set the appropriate pin to the OFF state.
4. You'll need to add additional positions to any of the arrays that you're still using (e.g. currentState, currentPeriod, currentTick, originalPeriod), but you can remove MAX_POSITION and currentPosition entirely since you're not tracking those for matrix pins.
5. Change `MAX_SUB_ADDRESS` to however many pins you end up using (24, 12, &c.), and add the appropriate number of blocks to the `tick()` function to handle all the pins.  You *may* be able to just replace all those blocks with a for loop instead (I wrote them all out explicitly as a bit of paranoia around making `tick` as short-running as possible).
6. Profit??? 😜 

Like I said, I'm probably missing a couple steps, but hopefully that sets you off in the right direction.  Can't wait to see it working!

SammyIAm, 2019-05-02:
Oh, forgot to add:

Right now, each "tick" toggles the pin at half the note-period so that the frequency of the square wave is correct, and the ON duty cycle is 50%.  If you want to adjust pulse length, you can take a look at `ShiftRegister.cpp` for maybe an idea of how that would work (in there the velocity value is used to affect pulse-length).

Chong-McBong, 2019-05-02:
thanks Sammy, i'm just on my way to bed, but will read and absorb this tomorrow. 

i like the idea of trying it with standard code first. i will do some endurance testing on a couple of pins. even if i melt them, i have plenty. 24 is way more than i need :)
,

closed 2020-08-12 by Sammy1Am