samphonic/Moppy2 #68: School project: adding another floppy drive

DeJerre1, 2019-05-28:
Hello

I'm Jeroen, a Belgian student.

I have managed to make a single floppy drive play (old version of moppy). Now I want to add another drive, but it doesn't give me an option to set a channel on the second drive.

I tried setting up the newer version, as I think it will be more fit for the project, but I can't seem to get it working. Is there any way I can use the old code to do this?

Thanks a lot!

Jeroen

SammyIAm, 2019-05-28:
It's been quite a while since I did anything with the old version, but if I remember correctly the second drive will always be channel 2 (there's no options for changing it).  Your best bet will probably be sticking with a MIDI file that uses consecutive channels (e.g. 1,2,3) like the sample MIDI files, but you can also connect the second drive to e.g. pins 6&7 on the Arduino to assign it to channel 3.

Apologies if I've misunderstood the issue you're having though.  A screenshot of your configuration might be helpful if there's something there that's causing trouble.

(On the other hand, what's not working with the new version?  It might be easier to get that working instead)

DeJerre1, 2019-05-28:
First of all, thanks for the quick reaction!
To get the new version, I unzipped the moppy file and eventually got the image in attachement.
[image]

I also got the MoppyControlGUI, but when I click on my COM3 port, there is no adress in the top right corner...
[image]

What am I doing wrong? I am totally new to this programming stuff ;)

Thanks!

DeJerre1, 2019-05-28:
First of all, thanks for replying so quickly!

I unzipped the moppy file, but when I opened the code in Arduino I got this:
[image]

I also got the MoppyControlGUI, but when I choose my COM-port, there is no adress in the top right.
[image]
What am I doing wrong? I am totally new to this kind of stuff :)

Thanks

SammyIAm, 2019-05-28:
No address in the top right means the MoppyControlGUI isn't getting a response back from the code on the Arduino.  A couple things to check:

- Did you successfully upload the Arduino code to your board?  Were there any error messages in the console at the bottom of the Arduino IDE?  (Sometimes it likes to fail very quietly)
- When you reset the Arduino after uploading the code, does the floppy drive play a short set of notes?  On startup, the 2.0 Arduino code should play a few notes to make sure everything on the Arduino side is set up correctly.  If you're not getting those notes, either the code upload has failed, or the wiring is somehow incorrect (however if you have one drive working with the old code, the wiring is the same so it's more likely a code issue)

DeJerre1, 2019-05-28:
Right now, I got to the point where the floppy drives make the sound and I got the adress in the top corner:
[image]

Right now my second floppy plays, but my first one (attatched to 2 and 3 of the arduino) doesn't play.

How do I fix this?

Thanks a lot!

SammyIAm, 2019-05-28:
Looks like there might be a bug in the included Condition scripts.  Channels 1-4 should probably be `c>=0 && c<=3` since channels are 0-indexed.  Try using the "All Events" condition script and see if that works with both drives.

DeJerre1, 2019-05-28:
Both drives are now reacting, but they are stuttering for some reason. I tried to upload a movie of it, but the file isn't supported.

Can it be because of my power source? Right now, I use the USB from my computer to the arduino.

SammyIAm, 2019-05-28:
The movie would be helpful, "stuttering" can be a lot of things.

First of all, you definitely should *not* be using the USB from your computer to the Arduino to then also power the floppy drives.  The floppy drives should be powered from some sort of external 5v power source.  (The Arduino itself can be powered from the USB no problem though).

Make sure you have one of the odd numbered ground pins on the floppy drives connected to GND on the Arduino.

What MIDI file are you using?

DeJerre1, 2019-05-28:
I got this external power source. the + is now plugged into the Vn and the - is plugged into the GND, but when I unplug the USB it doesn't power the arduino and floppy drives.

Right now, I'm using this midi file, but all notes are in range:
https://freemidi.org/download3-13466-game-of-thrones-tv-themes

SammyIAm, 2019-05-29:
The + should be connected directly to the 5v pin on the floppy drive itself, and the - to the appropriate ground pin (again on the floppy drive itself).  You shouldn't attempt to power the floppy drives through the Arduino.

If you use any of the sample MIDI files included with Moppy, does the stuttering still happen?  It might be MIDI-file related.

DeJerre1, 2019-05-29:
Applying another power source made it work!

Your help is really appreciated! thanks!

Jeroen

closed 2019-06-09 by DeJerre1