samphonic/Moppy2 #69: Wont run ubuntu 18.04

jmurray1217, 2019-05-30:
Hey I am really interested in getting this project to work, but I am having some real issues. I downloaded project, extracted it and ran "gradle build" in terminal and it wont compile. I read on one of the issues that there was a .bat file to execute everything in the bin folder but when I downloaded it there wasnt a bin folder at all. What am I doing wrong and how can I get this to work.

SammyIAm, 2019-05-30:
If you're only interested in running Moppy, and not customizing and building it yourself, make sure you download from the "Releases" part of the GitHub project (which includes the bin folder) rather than just downloading the source code.

If you're running in Ubuntu, you'll want to use the other binary in the bin folder, not the .bat file (which is for Windows)

jmurray1217, 2019-05-30:
Okay thank i got the GUI to come up now. Now time to work on the arduino not working with it thanks

closed 2020-08-12 by Sammy1Am