samphonic/Moppy2 #70: Stepper motor not running

Sesquialtera, 2019-06-09:
Hi, I've attached all the wires to the right pins on the Floppy drive, and the light comes on when the Drive selects are connected, but when the direction pins are connected and I connect the step wires together, the head does not move, and it has plenty of room to move. The motor does work because when give power to the floppy drive, it retracts. Why doesn't the motor move when the step pins are repeatedly connected?
Thanks,
Sam

SammyIAm, 2019-06-11:
It's possible you have the step and direction pins swapped.  Try it the other way around.

If you're really having to discover pin pairs by trial and error, you could try manually moving the head out a ways (try to do it carefully so it doesn't break) and just connect sequential pairs of pins until the head moves (those will be the step pins).

Sesquialtera, 2019-06-11:
Thanks for your reply. I think I may have done something to the floppy drive because the processor gets very hot quickly and the head is just buzzing and needs me to push it to move. I’ll try and find another!

closed 2020-08-12 by Sammy1Am