samphonic/Moppy2 #71: Transferring MIDI to floppy drives

DeJerre1, 2019-06-09:
Hi there

Not really a problem, more of a request:

I am trying to understand how this project works. What proces does it go through from the MIDI file to the floppy drive to make the specific sounds? In other words: how do the floppy drives read the notes from the MIDI file? How is the note transfered into a certain motor speed?

I was also wondering if there is a more extended version of the Moppy code. The first Moppy code had a lot of information and actual 'coding-stuff' in there, but this one is in my opinion more simplified (lots of text, but not actual 'coding-stuff').

Thanks a lot!

SammyIAm, 2019-06-11:
Happy to help, but I'll address the questions in reverse:

Can you give me an example of the 'coding-stuff' you're referring to?  Generally I've been trying to go for more-simplified, but it sounds like maybe I went too far or missed something.  I've not been actively making many changes to the code, but would be interested in feedback so that I can improve things in the future.

At a high level here's the process the music goes through (I'm happy to answer more specific questions on any of these steps):

1. The MIDI file is loaded by a [Java MIDI Sequencer](https://github.com/SammyIAm/Moppy2/blob/master/Java/MoppyLib/src/main/java/com/moppy/core/midi/MoppyMIDISequencer.java), and [MoppyMIDIReceiverSender](https://github.com/SammyIAm/Moppy2/blob/master/Java/MoppyLib/src/main/java/com/moppy/core/midi/MoppyMIDIReceiverSender.java) is registered as a listener on that sequencer.
2. When the sequencer is started, it begins interpreting the MIDI file and firing off MIDI events to the receiver.  These events can include a lot of different things (pitch bending and the like), but for the sake of simplicity we'll just look at note-on and note-off events.
3. These events are fed through [various mappers](https://github.com/SammyIAm/Moppy2/tree/master/Java/MoppyLib/src/main/java/com/moppy/core/events/mapper) that convert the MIDI information (channel, note number, data bytes, &c.) into MoppyMessages (device id, sub address, note number, &c.).
4. The MoppyMessages are sent over serial (or UDP) to a receiving device (generally an Arduino).
5. The code on the Arduino uses [an array](https://github.com/SammyIAm/Moppy2/blob/master/Arduino/Moppy/src/MoppyInstruments/MoppyInstrument.h) to map note numbers to periods (inverse frequencies), and for note-on events stores this value in the `currentPeriod` array (as ticks instead of microseconds, but they're more or less equivalent).
6. Every tick, the Arduino code increments the `currentTick` value for each drive and if it's reached the desired tick based on `currentPeriod` then it toggles the step pin that's connected to the floppy drive.  **Basically**, the step pin on the floppy drive is turned on and off at the frequency of the desired note.
7. There's some additional work to track the head position so the code knows when to toggle the direction pin to move the head back the other way.

Hopefully that covers your questions, but happy to answer more specific ones (or at least point you to the relevant bits of code).

DeJerre1, 2019-06-11:
Thanks for the reply. I feel like I know a bit better how the whole project works.

So when I look at the old Moppy code, I see 'coding stuff' like this:
[image]
[image]

But when I look at the newer Moppy file, I see only things like this:
[image]
[image]

I really like the newer version, but I have to explain the project to my teachers using the code. So my question is: how can I explain the project using the newer code? if I can explain it like you did in your last message using the newer code to show the different steps, that would be great!

Thanks a lot!

DeJerre

SammyIAm, 2019-06-11:
Ah, okay, I gotcha.

Yeah, in an attempt to make the code more readable and modular, a lot of the functionality that was in one giant file in the first Moppy has been moved into several smaller files with discrete functions.  So for example, the `tick()` function and most of the code from that original file are now in [FloppyDrives.cpp](https://github.com/SammyIAm/Moppy2/blob/master/Arduino/Moppy/src/MoppyInstruments/FloppyDrives.cpp).  And the `instrument.setup()` call calls the code there.

I don't know what class this is for, how much background you have, or how in-depth you're meant to go on this assignment (so forgive me if any of this is stuff you already know), but the general goal with splitting up the single giant file into more smaller ones is *abstraction* of functionality.  The idea being that `MoppyCore.cpp` shouldn't care if you're using floppy drives, or stepper motors, or serial ports, or UDP.  All it needs to do is read the configuration, initialize `instrument` and `network`, and then do the `loop()`.  Likewise, the [MoppySerial.cpp](https://github.com/SammyIAm/Moppy2/blob/master/Arduino/Moppy/src/MoppyNetworks/MoppySerial.cpp) can just focus on reading the writing from the serial port and also doesn't care if you're using floppy drives or stepper motors.

A lot of people were trying to use the original code to drive other devices, and breaking it up this way makes it easier to add new types of instrument to the code base.

tl;dr: The code from the original Moppy should mostly be in `FloppyDrives.cpp` and `MoppySerial.cpp`

SammyIAm, 2019-06-11:
Also: Depending on the class, this could be an excellent opportunity for a small class diagram, or flow diagram 😉

DeJerre1, 2019-06-11:
Oh I see. That explains it ;)

I do not have background on any of this, so everything that has to do with coding is difficult for me (I take this class as a choice option: it has literally nothing to do with my current education). My teachers have little understanding of this project, so I should explain it in a way they understand what is happening.

Right now, I think I understand everything from the FloppyDrives.cpp., except this bit (only copied for one of the pins)
[image]

If you got some more time, is it possible to explain the MoppySerial.cpp a bit more? I can't really see the proces behind the code.

I can't thank you enough for the things you already have clarified for me.

SammyIAm, 2019-06-12:
Sure, happy to explain:

The snippet from FloppyDrives.cpp is the bit that actually tells the floppy drive's stepper motor to take a step.  Each tick it does the following:

1. Check to see if `currentPeriod` for each drive is > `0`. If it's 0, then we're not playing any notes on this drive and we're done.  If it's greater than 0...
2. Increment `currentTick` to track how many ticks have passed since the last time we toggled the pin.
3. If a number of ticks have passed equal or greater to the desired period in ticks, toggle the appropriate pin, and reset the `currentTick` count.

(There's some additional complications (that aren't strictly necessary to understand probably) here in that the period that we're toggling the pin is half the period of the note we want because the floppy drive takes a step only on the transition from ON to OFF, but not OFF to ON.  So we call `toggle()` at half the period to create a square wave with a frequency of the note we want to play.)

Serial explanation coming in a moment...

SammyIAm, 2019-06-12:
[This page](https://github.com/SammyIAm/Moppy2/wiki/Design-Docs) on the wiki will be useful in understanding MoppySerial.cpp.

Basically, every time `readMessages()` is called:
1. The code checks to see if there's bytes waiting to be read on the serial port (a message from the Java program on the PC).  We're using `messagePos` to track which byte of the message we're currently expecting to see next.  Each time we successfully read a byte from the message, we increment `messagePos`, and when we're done we reset it to `0` to look for the `START_BYTE` again.
2. If there are bytes, it checks to see that the first byte is the `START_BYTE`, and if it's not we just ignore it.  We're doing this so that if some weird nonsense ends up on the serial line, or we accidentally miss a byte somewhere we can safely get back to the beginning of a message.
3. Once we're satisfied that we have the beginning of the message, work through the rest of the message bytes one at a time until we have enough information to either call the `systemHandler` or `deviceHandler` with the message data.  These functions are implemented in FloppyDrives.cpp (or whatever instrument is being used)

DeJerre1, 2019-06-14:
Thanks, this clarifies a lot!

I got two more questions if you don't mind:

1) What exactly does the TimerOne library do?

2) What program do you recommend using for editing the MIDI-files to get in range of the drives? Right now, I am using Sekaiju, but some MIDI-files get 'slowed down' when I play them on the floppy drives.

SammyIAm, 2019-07-02:
Wow, sorry, went out of town for a bit and had limited Internet access, just now finally catching up on emails.  This may be too late for your project, but hopefully still useful:

1. The TimerOne library makes it easier to use the Arduino's hardware interrupts (you can use them without the library, but it's a bit complicated so I used the library to make it easier).  Hardware interrupts basically tell the MCU to execute a particular piece of code periodically, interrupting anything that currently might be going on.  So in the case of the Moppy code, the code is continually polling and reading from the serial port in an infinite loop, and the interrupt routine calls the `tick()` function every 40 microseconds.  The execution of reading from the serial port will be interrupted/paused until `tick()` finishes.  This is why it's important that `tick()` execute very quickly, so that there's still time for serial reading.  The reason I used interrupts in this case is that because `tick()` is actually responsible for moving the drive head at a very specific frequency, it's important that it execute consistently even if there is more or less serial data to read.

2. I've had a bit of luck with Anvil Studio.  I already had Sibelius for some other arranging I was doing, so I used that for most of my floppy drive music, though it's pretty overkill for the most part.

closed 2020-08-12 by Sammy1Am