samphonic/Moppy2 #72: netbeans does not open the option to connect more floppy

aayalasalgado, 2019-06-30:
I have managed to connect a single floppy and it works very well. when you open the window to connect the arduino com, it does not open the right side where you can add more floppy. download the moppy2 files and I do not see the clear instructions as it was in the first pages of this project. I'm using win10 and the moppycontrolgui does not open. Until that instruction of moppy 2 I stayed. thanks for your help.

SammyIAm, 2019-07-02:
I'm not sure I'm following, sorry.  Is there any chance you could share a screen shot of what you're seeing (or maybe a red circle where you're expecting to see something that you're not).

aayalasalgado, 2019-07-06:
Gracias por contestar!  Ya puse a funcionar un solo floppy con moppy
original.  Mi problema es que no domino java y no puedo encontrar los pasos
a seguir para instalar los archivos de moppy 2.  No se si en win 10 hay
problemas para usar estos archivos.  Le envío una foto donde puede ver la
ventana que abre netbeans, pero al lado derecho de la ventana no aparece
los casilleros para añadir mas floppy.  Esta es mi duda y si pueden
ayudarme se lo agradecería mucho.  Quiero usar su proyecto en la escuela,
pues se que nadie lo ha visto anteriormente.  Le escribo en español pues no
domino el ingles y perdóneme por eso. Gracias!

SammyIAm, 2019-07-07:
Lo ciento que no me recuerdo mucho español.  😛  3 years in high school, but it's pretty much all forgotten.  Hopefully this translates okay:

- Make sure you download Moppy 2 from the [Releases](https://github.com/SammyIAm/Moppy2/releases) page.
- In Windows 10, right-click on `MoppyControlGUI-2.0.1.zip`, select 'properties', and tick the "Unblock" checkbox.
- Unzip `MoppyControlGUI-2.0.1.zip` to a folder.
- Open / run `MoppyControlGUI.bat` , and it should open Moppy2.  No additional installation needed.
- If this does not work, open a command prompt (Win+R, then `cmd`) and run `MoppyControlGUI.bat` from there and it will display any error messages.  Tambien there is a file called "moppy_control_gui.log" created in that folder that will have errors if it is not working.

The `MoppyArduino-2.0.1.zip` file should work the same way it did with the original Moppy.  Just open in the Arduino IDE and upload.

Buenos suerte!  Let me know if that helps.

closed 2020-08-12 by Sammy1Am