samphonic/Moppy2 #73: Unable to build Moppy2 with eclipse and Gradle

devfilmer, 2019-07-05:
Hello,

 I started with a clean install of Eclipse IDE 
Version: 2019-06 (4.12.0)
Build id: 20190614-1200
OS: Windows 7, v.6.1, x86_64 / win32
Java version: 12.0.1

Then imported moppy2 

However, I am unable to build moppy from the latest 2,0,1 branch due to numerous undefined methods. 

Here are a few.
Description	Resource	Path	Location	Type
The method builder() is undefined for the type DeviceDescriptor	NetworkManager.java	/MoppyControlGUI/src/main/java/com/moppy/control	line 86	Java Problem
The method getConditionChoice() is undefined for the type MoppyConfig.MIDIScriptMapperConfig	MapperPanel.java	/MoppyControlGUI/src/main/java/com/moppy/control/gui/mapperpanel	line 63	Java Problem
The method getConditionCustomScript() is undefined for the type MoppyConfig.MIDIScriptMapperConfig	MapperPanel.java	/MoppyControlGUI/src/main/java/com/moppy/control/gui/mapperpanel	line 438	Java Problem

If this is a gradle configuration issue (i.e. my error) I apologize.
Any help would be appreciated.

SammyIAm, 2019-07-10:
It looks like what all those methods have in common is that they should be generated/injected by Lombok.  Offhand I don't remember/know the process for getting Lombok setup in an Eclipse project, but I think that's the direction you should head to resolve this.

Sammy1Am, 2020-06-10:
Ah, DeviceGUI is a Java UI simulator of a floppy drive.  I haven't kept it updated so I'm not 100% sure it still works, but I used it for testing the Java code when I didn't have an Arduino and floppy drives handy.

Sammy1Am, 2020-08-12:
Leaving this open for now as a reminder to improve the documentation.

devfilmer, 2020-08-12:
Cool
Glad to see you working on moppy.

I was able to set up moppy to work with a raspberry pi and JAVA.
Not a huge accomplishment.. but it works.
Ironically.. I believe I accidentally wiped the the raspberry pi card while
working on another project.

I'll have to go back and install it again and then get back to figuring out
the command line stuff.

Thank you for all your hard work on Moppy.

I am a member of a maker space and we made this 2 years ago for Maker Faire
Orlando. Laser cut wood in a box which was intended to increase the sound
of the drives. (Loudness needs to still be increased)
My current goal to make it a stand alone exhibit tied to buttons with
various midi files.

Without you this would not exist.

Thank you,

A.

Sammy1Am, 2020-08-12:
Glad you got it working!  Yeah, finally dusting off the code and putting in some much needed updates.

There's a ticket (#67) related to using buttons on the Arduino for input, and one for playlists (#91).  They're low on my own to-do list (the primary goal is making things easier to get people up and running, and these might just complicate things), but if you do end up writing any code for those maybe we can pull it into the repository (I know a lot of people like having a playlist option).