samphonic/Moppy2 #76: ControlGUI

JE521, 2019-08-21:
Hi, I know moppy is really old, but i was interested. I am on pc. I followed the directions on the wiki, but when I am supposed to open the "MoppyControlGUI.bat", (I was given the prompt to agree to run it) , it opens for a millisecond and then closes. The Hardware and arduino set up worked fine, and i am using moppy 2.0.1. If you could help that would be much appreciated. Thank you

SammyIAm, 2019-08-21:
Replying by email so excuse the brevity (I can try more later if this
doesn't help):

Instead of double clicking the .bat, try opening the command line
(Start>run>cmd) and running the bat from there.  It'll show any errors that
occur hopefully.

Also look around in the Moppy folders for log files that might have errors.

Maybe try before unzipping, right clicking, properties, and check
"unblock", then unzip.

If it's still not working, or there's an error, let me know and I'll
respond at length later.

Good luck!

JE521, 2019-08-21:
Thank you for responding. I copied the text contents of the .bat file and pasted it in command prompt. I hop thats what you meant. but it did the same thing. I also unblocked it. when i clicked on it, it would pull up a window and had the options "extract all", "Run", and "Cancel". I clicked run, did the same thing. I extracted all and it just made another duplicate moppycontrolGUI folder. 

I also noticed in the wiki, it has no mention of netbeans, is netbeans not required anymore?

SammyIAm, 2019-08-28:
Yup, 5.25" drives work just fine (though some of them may have less tracks, so you'll need to adjust the maximum position for those drives in the Arduino code (to 79 instead of 158)).

You can add up to 9 floppies by just connecting them to the next available pair of pins (e.g. 2&3, 4&5, &c.).

JE521, 2019-08-28:
Do you have a pinout diagram by chance? I can't seem to find it in its manual.

SammyIAm, 2019-08-29:
5.25" drives, I believe, have the same pin ordering as the 3.5" drives, just a different connector type.

JE521, 2019-08-29:
I am working on the connections, but how do I change the max position for any of the drives. I checked the code and couldn't find any variable for the max position. Also is it ok to connect all of the drive's ground pins to the same ground pin on the arduino. I doubt it but I don't see much different ground pins on the arduino.

JE521, 2019-08-29:
the drive does not work, is it possible to control the stepper by directly connecting it to the arduino. I looked online and in your previous issues but can not find anything. Thanks again so much for your guidance.

JE521, 2019-08-29:
I also can not seem to run the working floppy on any of the other pins for additional drives, i was checking to see if it worked and it didn't.

SammyIAm, 2019-09-05:
GitHub has a [useful search function](https://github.com/SammyIAm/Moppy2/search?q=max+position&unscoped_q=max+position) to find things in a repository.

It's not possible to control the stepper motor directly from the Arduino (there is some circuitry required to drive stepper motors, but it's conveniently included in the floppy drive hardware which is why floppy drives work so well).  It *is* possible to drive stepper motors directly using Moppy, but you'd need a driver board connected to the Arduino to do so.

Make sure the MIDI file you're using has enough tracks for the other pins to have things happening on them, and make sure your mapping settings in the GUI are sending events for all the MIDI channels.

DarianKimberly, 2019-09-08:
Is there a linux version or will I need to completely rewrite the code in a compatible format?

SammyIAm, 2019-09-08:
@DarianKimberly , Moppy's written in Java so you're good to go pretty much anywhere.  In the `bin` directory there's a file named `MoppyControlGUI` that *should* (I haven't actually tested) allow you to easily run the ControlGUI.  If that script doesn't work out of the box, it should only require minor adjustments, or you should be able to run `MoppyControlGUI-2.0.1.jar` directly as long as you set the classpath correctly.

DarianKimberly, 2019-09-08:
Not sure how to set the classpath. New to this.

SammyIAm, 2019-09-08:
Did the script not work?  What error are you seeing when you run it?

I'm happy to try to help answer specific questions; but because there are a ton of different variables to a given setup and Moppy generally just uses pretty standard libraries, you'll probably have just as much (or better) luck on Google for a lot of the generic getting-Java-working-on-my-system questions.

DarianKimberly, 2019-09-08:
Just trying to figure out if I can control my arduino from my Raspbian
setup as it is linux based. Not sure if it is compatible. the controller
program is a windows batch file, and im not sure how to convert it into
something linux friendly.

DarianKimberly, 2019-09-08:
The reason behind this is that I want to make a portable version of the
moppy built that i can control wirelessly through a raspberry pi zero w
connected to my uno and VNC viewer.

SammyIAm, 2019-09-08:
So, just to clarify again, the controller program is a `.jar` file.  There is a Windows .bat file included for running on Windows machines, **and a shell script** (the file I mentioned before in the `bin` directory) for running on Linux/Unix.

Try the shell script.

DarianKimberly, 2019-09-08:
I tried and it just opens as a text file

DarianKimberly, 2019-09-08:
Update. I manually set it to executable (for some reason this is not a
default), and it opens for a brief moment, then just closes.

SammyIAm, 2019-09-08:
Try executing it from a terminal.  It should output any errors in encounters (you can also look for `*.log` files in the Moppy directories).

If I had to guess though, it's probably a `JAVA_HOME` environment variable issue.  Make sure that's set up correctly.  (But try running from a terminal to look for the error first, it might be something else)

DarianKimberly, 2019-09-08:
Again, no dice. I already tried executing it from the terminal.

SammyIAm, 2019-09-08:
What was the output when you tried to run it?  (I can't help if I don't know what issues you're running into)

DarianKimberly, 2019-09-08:
Executing in terminal results in a terminal that opens for a split second
and immediately closes. Nothing else happens.

JE521, 2019-09-09:
you could try the JAVA_HOME thing, just scroll up to beginning.

JE521, 2019-09-09:
I think that the pin problem might be due to my use of an arduino mega 2560 instead of an uno. I will try to purchase an uno if there is no way to do this with my board. Also I will be less active on github because of school.

DarianKimberly, 2019-09-09:
What do you mean by that?

SammyIAm, 2019-09-09:
@JE521 , The Arduino Mega should work fine for the first set of pins (roughly 2-19 or so).  Can you maybe send a screenshot of your ControlGUI, just in case something with the mappers is preventing channels higher than 1 from being sent to the drives?

No rush of course, enjoy (?) school.  ;P

SammyIAm, 2019-09-09:
@DarianKimberly .  JE521 is referring to the original topic of this issue which he created.

Generally it's best to create a new issue if you're asking for help with an unrelated problem.

DarianKimberly, 2019-09-09:
The problem is getting the GUI program to open. It just opens a command
line for an instant, then immediately closes.

SammyIAm, 2019-09-09:
@DarianKimberly : I was talking to JE521, not you.  Why don't you open up a new issue and we'll see if we can't get you sorted out over there.

DarianKimberly, 2019-09-09:
Sorry. My bad. It emails me with every reply.

JE521, 2019-09-13:
[image]

JE521, 2019-09-13:
Also I have a driver board for my stepper.

SammyIAm, 2019-09-13:
Alright, so... this might seem like a silly suggestion, but I figure we should just make sure...  The MIDI file you have open is called `KirbysTheme (1 floppy).mid` which I assume has been written for just one floppy drive (and therefore probably doesn't contain any notes for any of the other drives.

As something to try though, you can click "Add Mapper" and set up the mappers like this:

| | | | |
|---|---|---|---|
|`n>22 && n<67`|`1`|`1`|`n`|
|`n>22 && n<67`|`1`|`2`|`n`|

That will basically take any note that comes in and assign it to SubAddress 1 (first row) and SubAddress 2 (second row).  That'll make the first two floppy drives play the same notes, but could help testing your second drive.

Regarding the stepper motor, as long as the board is sufficiently similar to the L298N board, you should be able to comment out the two FloppyDrives lines in `MoppyCore.cpp` and uncomment the L298N lines.  The L298N code was contributed by someone else so I don't know the nitty gritty, but if it doesn't work out of the box it should work with some small modifications.

JE521, 2019-09-13:
That sounds like a good idea, but the 4,5 pins and so on don't work, I try restarting arduino on those pins and floppy wont play startup sound. Also the subaddress says play for 1 floppy because I set that when I saw the issue you forwarded me.

JE521, 2019-09-13:
Also the 5.5" floppy doesn't work, so I am trying to tap into its stepper motor. And I am getting more 3.25" floppies soon.

SammyIAm, 2019-09-13:
- The startup sound *only* plays on the first drive.  (You could change it in the [Arduino code here](https://github.com/SammyIAm/Moppy2/blob/master/Arduino/Moppy/src/MoppyInstruments/FloppyDrives.cpp#L85) if you wanted to)
- The **MIDI file** says "1 floppy" (try a different MIDI file!)
- You should be able to figure out how to get the 5.25" floppy working by manually grounding pins to see which ones control the stepping, &c.  I guess it's possible the drive itself is dead and then you could just bypass it and control the motor directly.

JE521, 2019-09-13:
I hooked up the right pins to the 2,3 pins on arduino. it wont move.

JE521, 2020-01-03:
I am sorry i cut off this issue without notice, my computer broke and I was busy with school. I don't have the time for this kind of project. Thanks again for your help

DarianKimberly, 2020-01-03:
Is fine. Education is important. I quit my job because they asked me if I
valued education or my job more.

SammyIAm, 2020-02-10:
Closing for now, good luck with school.  Feel free to re-open if you try again and have the same issue.

closed 2020-02-10 by SammyIAm