samphonic/Moppy2 #77: Midi (round robin with less than 8 drives)

johndfoss, 2019-08-31:
Hi!

First of all, thanks for the great work!

I am new to this, so bare with me :-)

I am using Moppy on a Windows PC, connecting to an Arduino Uno with two 3.5" drives connected.
When I play single notes on my MIDI Keyboard that is connected to the PC I get perfect output on Moppy.
When I try to play two notes simultaneously I start running into problems.
As far as I understand it should be possible to play several different notes at the same time (drive A plays note 1, drive B plays note 2, ...) by using Round Robin. When I try this with just two drives connected there are a number of problems. The drives do not reliably start and stop when the keys are pressed and released, I cannot reliably reproduce polyphony although it does work for brief periods in between.
Is there any way to change the Round Robin Sub-Address mode to just use two drives or does it require eight drives?
(Unfortunately I haven't been able to find the code that controls the Round Robin mode anywhere and would be grateful for any pointers).

Thanks!

SammyIAm, 2019-09-05:
Hi there!

Sorry for the delay in responding, I'll tackle your questions in reverse:

- As for the round-robin code, I'll point you to GitHub's [useful search feature](https://github.com/SammyIAm/Moppy2/search?q=round+robin&unscoped_q=round+robin).  But to change from 8 drives to 2, all you need to do is change the SubAddress script from `nextRoundRobinSubAddress(8)` to `nextRoundRobinSubAddress(2)` and it should use 2 drives instead of 8.
- The drives not starting reliably is most likely because sometimes the note is started on e.g. drive number 5 (which you don't have), so making the adjustment above to use 2 drives instead should hopefully fix that.
- Once you have round robin set to use 2 drives, if you're still having trouble with notes not stopping reliably then there may be a bug somewhere in the code.  The way the round robin code works is it assigns each new note-on event to the next drive (1, 2, 3, etc.) and after it assigns a note to the last drive (as specified by the argument to `nextRoundRobinSubAddress(x)`) it wraps back to 1.  It also tries to keep track of which notes are currently being played so that when a note off event comes in, we can search for the note and turn it off appropriately.
It's definitely possible that there's an error in the logic for locating and turning off notes (or there's some ordering issue that can cause notes to leak that I hadn't thought of).  If you can reliably reproduce a stuck note, let me know how you're able to do it and perhaps we can debug the code (I don't have my drives with me so I can't test myself).

Hopefully that's helpful!

johndfoss, 2019-09-12:
Quite helpful thanks :-)

Regarding bullet point 1: do you mean change from nextRoundRobinSubAddress(8) -> nextRoundRobinSubAddress(2) in the GUI or in the actual source code?

Sammy1Am, 2020-07-06:
Closing due to lack of activity.

closed 2020-07-06 by Sammy1Am