samphonic/Moppy2 #78: Moppy software issues.

DinukaShehan, 2019-10-01:
Hi Sammy,
It is a nice project that you are created.
Firstly I want thank for it.

I downloaded new moppy files from your github.
I have installed java 7u80 and netbeans 7.1.2.
My laptop is x64 bit.
Problem are i couldn't find rxtx serial drivers.
Couldn't find .bat file when i open a project from netbeans.
I download moppydesk from somewhere around.
When i open it from net beans it shows a error in JavaUI.

Have i jumbled the steps.
Please direct me to the success.
Thank you

SammyIAm, 2019-10-01:
Hi there!

A few things to double check:

- It sounds like you might be looking at instructions for the old version of Moppy, take a look at the [wiki for 2.0](https://github.com/SammyIAm/Moppy2/wiki)
- Make sure you're downloading the code from the [releases page](https://github.com/SammyIAm/Moppy2/releases).
- NetBeans is no longer required to run Moppy (you only need it if you're going to be modifying code).
- RXTX is not needed.
- You'll find the `.bat` file in the `bin` directory after you unzip the file downloaded from the releases page.

DinukaShehan, 2019-10-02:
Thank you for reply.
I will try your instructions and inform you whether it work or not!

DinukaShehan, 2019-10-03:
What is the suitable java software

DinukaShehan, 2019-10-03:
Sorry I should repeat my question.
When I open .bat file;just blink a cmd screen and nothing happens after that.
No MoppyDesk window.
My laptop is i3,windows 7 64bit.
Why this happens?
Please kindly explain this.
Thank you

SammyIAm, 2019-10-03:
My first guess is that that `JAVA_HOME` isn't set correctly (just like the original problem in issue #76 ).  You can try making sure that's set correctly first.

Otherwise, run the `.bat` file from a command prompt window (i.e. open the command prompt *first*, navigate to the bin folder, and then type in the .bat file to run it), and it should hopefully show you any errors it's running into.

DinukaShehan, 2019-10-04:
[image]

DinukaShehan, 2019-10-04:
[image]

DinukaShehan, 2019-10-04:
error with my java software?

DinukaShehan, 2019-10-04:
java software valueshouldC:\Program Files\Java\jdk1.7.0_80\bin or C:\Program Files\Java\jdk1.7.0_80

SammyIAm, 2019-10-04:
[A quick Google search](https://www.google.com/search?q=java+unsupportedclassversionerror) reveals that this error indicates mismatch in the compiled class files and the JVM running them.  It appears that like I said before, Java 8 is the minimum version required.  Just install that, make sure JAVA_HOME points to the Java 8 jdk, and then you should be good to go.

DinukaShehan, 2019-10-04:
[image]

DinukaShehan, 2019-10-04:
[image]

DinukaShehan, 2019-10-04:
If I connect all the hardware correctly, after that opens the moppy control window.
How can i control number of floppy drives,tones which to play each drive?
It will automatically connect with Arduino?

SammyIAm, 2019-10-04:
If you're only using a single Arduino, each MIDI channel will be assigned to one of the drives.  That is, channel 1 to the first drive, channel 2 to the second, etc.

If for some reason you want to try to map things on-the-fly, you can use the mapper scripts (at the bottom of the GUI window) to map different notes to each drive.  Each script is a Javascript:

- Condition filters which notes should be considered for that mapping row (`true` = include, `false` = don't)
- Device Address will always be 1 if you only have 1 Arduino
- Sub Address will be 1, 2, or 3 depending on which of the drives you want to send that note to.
- You can leave Note as `n` if you don't want to change the notes at all, but you can also use that script to raise or lower the note pitch (e.g. if the notes in the MIDI are too high).

Probably just stick with 3-channel MIDI files to start, and once you have that working you can mess around with the scripts.

DinukaShehan, 2019-10-07:
Thanks sammy
I play music with only one drive.
Little sound comes with it.i think this floppy drive is not noisey enough.
I get 3 floppy drives and one Arduino uno.
How can i play music with this.
Hardware configuration is ok.

Moppy software how to select notes and add 3 floppy drives

SammyIAm, 2019-10-07:
Floppy drives are (un)fortunately not supposed to be very loud, so that's just how things are.  I've seen some people make them a little louder with passive amplification (like putting them on a hollow box or attaching plastic cups to them).

The default mapping and Arduino software should work fine for up to 9 drives.  The Just connect the second drive to pins 4&5, the third to 6&7, etc.  Make sure the MIDI file you're using has at least 3 channels.

DinukaShehan, 2019-10-08:
https://youtu.be/hk5reJCAX_I

DinukaShehan, 2019-10-08:
c+1?what is it.
If i use 3 floppy drives; all of them play same note or different?

SammyIAm, 2019-10-09:
Nice!  Looks like you got it working.

There's a [little information on the wiki](https://github.com/SammyIAm/Moppy2/wiki/Scripting), but `c` is the channel defined in the MIDI file (from 0 to 15), so `c+1` will map those channels to 1 though 16.

If you use 3 floppy drives without changing the mapping, each drive will play a different MIDI channel.

DinukaShehan, 2019-10-09:
Can I use nema 23 stepper motors with stepper motor drivers without using floppy drives

SammyIAm, 2019-10-09:
I don't have any experience with those.  In theory I'm sure it's possible, I just don't know what you'd need to modify.

DinukaShehan, 2019-11-06:
I have tried with nema 23 stepper motors with tb6600 stepper drivers.
It worked fine.
I will send you a video clip later.
Thank you for great help sam

SammyIAm, 2019-11-06:
Awesome!  Glad you got it working, look forward to the video :)

closed 2020-08-12 by Sammy1Am

HunterG6700, 2021-02-12:
never sent that video lol