samphonic/Moppy2 #80: RoundRobin and 16 Drives

DiMoSe, 2019-10-31:
I'm trying to play 16 drives on an Arduino Due, however the program as it is presented in the release is not able to do it. I think it's possible based on [this issue](https://github.com/SammyIAm/Moppy2/issues/43) and [this one](https://github.com/SammyIAm/Moppy2/issues/45) I found. Unfortunately I have no idea what modifications @THLCX made to the code since his file is no longer available and I don't know how I would build the GUI app from the source code to make it able to do RoundRobin in 16 drives.

Can anyone help me? (◕︵◕)

SammyIAm, 2019-10-31:
I can help! 🦸‍♂️

The first good news is that no modifications are necessary to to GUI code to support any number of drives (or Arduinos).  All you need to do is change the `8` in the round robin Javascript script to a `16` and you're all set!

As for the Arduino code, one easy change is:

- In `MoppyConfig.h` change `MAX_SUB_ADDRESS` from `8` to `16`.

and then there's only a few other changes in `FloppyDrives.cpp` (I can't guarantee these are all of them, but should be the main ones at least):

- On line 13, change the `8` to a `16`.
- On lines 31, 34, 43, 46, and 49 add additional elements to the array so that instead of the max element being index 9, it's index `16` (e.g. there should be 17 total elements since the first one is 0)
(Noticing a pattern? 😛 )
- On line 39, extend the array from `19` to `33`.
- Add more `pinMode` lines around line 71.
- Finally, you'll need to add more `if` statements after the `if` statement line 232 all the way up to pin 16.

That should more or less do it, but like I said I don't know for *sure* if that's all the places.  Should at least get you on the way though.

Let me know how it goes and if you have any additional questions.

DiMoSe, 2019-11-06:
Thanks a lot!! I have one question so far. On line 39 the first two elements of the array have a value of `0`, why is that and should the values that I add all be `LOW`?

SammyIAm, 2019-11-06:
The first two values are `0` because they aren't used, everything you add should indeed be `LOW`.  Pins 0 and 1 on the Arduino can't be used for controlling drives (easily anyway) because it would interfere with serial communication.  To keep from having to do `currentState[pinNumber-2]`, I just added the unused ones to trade a little memory to save on CPU cycles.

Huh, well, I counted several times, and the array on line 39 *definitely* has a length of 19, but it **  should** actually have a length of 20 (pin 19, the last pin initialized for 9-drive operation would be the **20th** pin).  So that's a mistake on my part.

For 16 drives, you'll need to use pins 2-33.  So the **index** of the last array element should be 33, and therefore the array will need to have a length of 34.

closed 2020-08-12 by Sammy1Am