samphonic/Moppy2 #81: How to use 5.25" drives?

adaringdesign, 2019-11-17:
I have 3 5.25 drives here but I can't get them to do anything. I've wired them exactly the same as the 3.5 drives that I have but still nothing. Am I missing something?

SammyIAm, 2019-11-18:
Hmm, in **_theory_** they're the same, but it does seem like the 5.25" drives can be a bit tricky for some people.  As far as troubleshooting, I'd say:

- Hook it up to a regular PC power supply (if you haven't already).  I know that the 3.5" drives only need 5v, but the larger ones might use 12v.
- The pins on the connector should still be wired in the same pattern, that is the odd pins are ground and the even pins do stuff.  You can try confirming which row of pins is the odd/ground pins if you have a multimeter.
- Using a small conductor (wire, paper clip, pliers, small coin, etc.) connect pairs of pins on the connector (i.e. 1&2, 3&4, 5&6, etc.) until you manage to get the drive's "activity" light to turn on.  This should indicate that the drive is now "selected" and should respond to step signals.
- Now, unfortunately, you'll have to have located the DIR(ection) pin and grounded it (or connected it to its matching odd-pin) before the STEP pin will actually move the head forward.  Again, in **theory** they should be next to eachother, so maybe just try connecting pairs of even/odd pins that are next to eachother until the drive head jitters a little.

Basically what you're aiming to do is manually locate the pin that causes the drive head to move, and the pin that controls the direction of the drive head.  If you manage to find those, you can then be sure that you've connected them properly to the Arduino.

adaringdesign, 2019-11-18:
Thanks, I'll try that and post my findings!

I did find [this](enable_pin) however and while it does mention pin12 being grounded, it also mentions pin16 set to high on the arduino.

adaringdesign, 2019-11-19:
I figured it out, pin10 (select 0) needs to be grounded for the drive to become active!

SammyIAm, 2019-11-19:
Nice!  Glad you got it figured out (darn weird 5.25" drives :P )

closed 2019-11-23 by adaringdesign