samphonic/Moppy2 #83: Adding second floppy problem

Exy25, 2019-11-30:
Hi Greeting from Czech Republic.
I tried to add second floppy but when  connected it and uploaded the arduino code
start up sound maked only first floppy, the second moved just a little bit and then it did nothing.
So I ask you is it a software bug or some wiring because I thing that I connected it all right.
first floppy to 2 and 3 pin and second floppy to 4 and 5. GND is connected to.
Do I have to change some code or configurations, or it is driver bug or something.

Please reply. I will be so thankful to you.

SammyIAm, 2019-11-30:
It sounds you have everything hooked up correctly.  The startup sound only
plays on the first drive by design; the second drive shouldn't do anything
on startup, but should still play music with the right MIDI file.

Sam

SammyIAm, 2020-02-10:
Closing due to inactivity; feel free to reopen if the issue persists.

closed 2020-02-10 by SammyIAm