samphonic/Moppy2 #84: Moppy control gui drawn oddly on linux mint 19.3 (based on ubuntu 18.04

unorthodoxcoder, 2020-01-29:
I have been messing with moppy on Linux mint 19.3 when i noticed the control gui seems to be visually corrupt (Assuming that the GUI is meant to look the same across platforms here) 
Mint
[image]
Windows
[image]

when i try to compile the GUI myself on the mint system (via net beans 10.0) to try and pin this down i receive a Project does not have a main class set error as this is my first time working with java i am at a loss on how to proceed

SammyIAm, 2020-01-29:
It actually looks mostly okay to me (unless I'm missing something major).  If I recall correctly, Java has some options for forcing a certain look, but I chose to use the native system UI elements (hence rounded vs square buttons, etc.) The only bits that seem off are the missing symbols for stop and play (probably a font thing), and that there's some numbers above the sliders in Mint but not Windows.

- A quick Google seems to suggest you can use `java -Dswing.plaf.metal.controlFont=Tahoma -Dswing.plaf.metal.userFont=Tahoma` to change/force the default font to something else to get those symbols to appear (or maybe your system default font if that doesn't work).  There's not much to be done here fix-wise aside from maybe bundle fonts I guess (but then it would look very weird indeed).
- I don't know why there are suddenly numbers above the sliders in Mint but not Windows, but there's probably a way to disable them (or at least make space so that they look okay).  Now that I'm looking though, the volume slider seems to either be invisible or the same color as the background which is a bit weird.

I'll leave this issue open and maybe take a stab changing some of these next time I'm in the code (though I don't have a Mint system so I might have some trouble recreating them).

As for running the project in NetBeans, there's somewhere to configure NetBeans to tell it basically "If I say to run this project, start with the `main` method from X class".  In the case of the MoppyControlGUI, the class in question is the `MoppyControlGUI` class.  I don't have NetBeans installed so I don't know offhand where to change that configuration, but Google should help again there too.  You should also be able to right-click inside the `MoppyControlGUI.java` source code and see something like "Run this file..." or "Run from here..." which should also start the application.

SammyIAm, 2020-01-29:
Ah, actually, re-reading my response gave me an idea and sure enough the "LookAndFeel" of the application is set [on line 71 here](https://github.com/SammyIAm/Moppy2/blob/master/Java/MoppyControlGUI/src/main/java/com/moppy/control/MoppyControlGUI.java#L71).  You can change `getSystemLookAndFeelClassName()` to something else like `getCrossPlatformLookAndFeelClassName()` and see if that makes things less weird (though it'll probably be subjectively uglier).  There's (maybe) more info on that [here](https://docs.oracle.com/javase/8/docs/api/javax/swing/UIManager.html).

unorthodoxcoder, 2020-01-31:
Hmm, still no luck running via Netbeans. I hate to trouble you but if you could provide step by step import instructions, I would appreciate that greatly. I suspect that I have screwed something up there seeing as Netbeans will not even allow me to open the class file let alone run from there. However looking at the code here on Github i  do see what you mean in regards to line 71 thank you for your help.

SammyIAm, 2020-01-31:
Well, I don't have NetBeans installed at the moment so I'm unable to walk through the process myself, but step-by-step it should be something like:

1. Open the MoppyControlGUI project in NetBeans.
2. Press "Run".

But since that's not working it sounds like something unexpected has gone wrong.  A couple things to check to help you troubleshoot:

- Do you have the [Gradle support plugin](http://plugins.netbeans.org/plugin/44510/gradle-support) installed?  You can install it from inside NetBeans in the plugins area.  The projects use Gradle to build, so this is a requirement.
- What steps are you following to open the project in NetBeans?  Are you getting any unusual prompts or errors along the way?

<hr>

> Netbeans will not even allow me to open the class file

You mentioned you were new to Java; I'm not sure what language(s) you're coming from but a quick Java primer on files: Java's source code is written / stored in `*.java` files.  Generally one Java "class" (the same sort of OOP class concept that that other languages use) is defined in a `*.java` file of the same name.  Java source is compiled into `*.class` files, one per `*.java` file, which contain the Java bytecode (they're binary files; you can't edit them in a text editor).

So after opening the project in NetBeans, the **file** that you'll open is the `MoppyControlGUI.java` file, which contains the MoppyControlGUI "class" (and is compiled into `MoppyControlGUI.class`, but you're not concerned with that file now).  More importantly, `MoppyControlGUI.java` contains the `main()` method which is the entry point for the application.

You should be able to right-click inside the `MoppyControlGUI.java` and select "Run" or "Run this file" as long as the project has been opened in NetBeans correctly.