samphonic/Moppy2 #86: NetBeans Error, HELP ME !!! Please

krashone, 2020-02-09:
Hi, could you help me?
NetBeans gives me the following error.

**A fatal error has been detected by the Java Runtime Environment:
 EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000006af4862e, pid=5048, tid=11084

JRE version: Java(TM) SE Runtime Environment (13.0.2+8) (build 13.0.2+8)
Java VM: Java HotSpot(TM) 64-Bit Server VM (13.0.2+8, mixed mode, sharing, tiered, compressed oops, g1 gc, windows-amd64)
Problematic frame:
C  [libNRJavaSerial.dll+0x862e]

No core dump will be written. Minidumps are not enabled by default on client versions of Windows

An error report file with more information is saved as:
C:\Users\krash\3D Objects\Floppy Music\MoppyClassic-moppy-advanced\Java\MoppyDesk\hs_err_pid5048.log

If you would like to submit a bug report, please visit:
  http://bugreport.java.com/bugreport/crash.jsp
The crash happened outside the Java Virtual Machine in native code.
See problematic frame for where to report the bug.

C:\Users\krash\3D Objects\Floppy Music\MoppyClassic-moppy-advanced\Java\MoppyDesk\nbproject\build-impl.xml:1330: The following error occurred while executing this line:
C:\Users\krash\3D Objects\Floppy Music\MoppyClassic-moppy-advanced\Java\MoppyDesk\nbproject\build-impl.xml:968: Java returned: 1
BUILD FAILED (total time: 28 seconds)**

How do I solve it?
Thank you

SammyIAm, 2020-02-09:
Make sure you're using the [latest release](https://github.com/SammyIAm/Moppy2/releases) from Moppy **2.0**.  It looks like you're trying to run the old MoppyClassic code which isn't supported anymore (largely in part because of the serial driver issue you've found)

closed 2020-02-09 by SammyIAm

krashone, 2020-02-09:
now gives me this mistake

JAVA_HOME="C:\Program Files\Java\jdk-13.0.2"
cd C:\Users\krash\3D Objects\Floppy Music\Moppy2-master\Java\MoppyControlGUI; C:\Users\krash\.gradle\wrapper\dists\gradle-4.10.2-bin\cghg6c4gf4vkiutgsab8yrnwv\gradle-4.10.2\bin\gradle --configure-on-demand -x check run
Configuration on demand is an incubating feature.
> Task :MoppyLib:compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':MoppyLib:compileJava'.
> java.lang.ExceptionInInitializerError

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/4.10.2/userguide/command_line_interface.html#sec:command_line_warnings

BUILD FAILED in 0s
1 actionable task: 1 executed